fdivitto/FabGL #117: [Feature] VGADirectController: ability to repeat same line twice

abelykh0, 2021-05-14:
I would like to use some modes when the same line is repeated twice, for example, 640x480 would become 640x240.

abelykh0, 2021-05-14:
See https://github.com/fdivitto/FabGL/pull/118

abelykh0, 2021-05-14:
My original idea wan actually not even call the drawScanline 2nd time. The draft pull request calls it again. It is more flexible, but probably will just cause confusion.

abelykh0, 2021-05-18:
The "ideal" API would be something like void IRAM_ATTR drawScanline(void* arg, uint8_t* dest, int scanLine, uint8_t* repeatLines). Then I set *repeatLines = 2;

i-am-shodan, 2021-08-17:
Is this what your after?
https://github.com/fdivitto/FabGL/pull/142

abelykh0, 2021-08-17:
> 
> 
> Is this what your after?
> #142

Not really, this is for a Sinclair Spectrum emulator, which you already helped me with previously in https://github.com/fdivitto/FabGL/issues/85. For example of intended use see https://www.youtube.com/watch?v=OEjMUaaSh-4 0:17 to 0:23 transition. At 0:17 it is repeating every line, but not at 0:23.

fdivitto, 2021-10-04:
VGADirectController allows you to set number of lines per callback using setScanlinesPerCallBack method:
```
    directCtrl.setDrawScanlineCallback(....
    directCtrl.setScanlinesPerCallBack(2);
    directCtrl.setResolution(....
```

abelykh0, 2021-10-04:
It doesn't work. Looks like it is expecting me to fill two lines inside the callback, not repeat the same line twice. The goal is to reduce time inside the callback as much as possible, with this look like I have to do extra copy of the same line. 
Maybe it is my misunderstanding, do you have an example for this?

fdivitto, 2021-10-05:
Yes, you should fill every line. 
To repeat the same line without fill it is necessary to dynamically change the DMA chain. Let me think about...

fdivitto, 2021-10-05:
This could be possible: a method could specify the scanlines that shares the same buffer. This method can be called **one time** or **every time** you need to change display configuration.

For example, to make scanline 1 the same of scanline 0, execute:
```
directCtrl.setScanlineBuffer(1, directCtrl.getScanlineBuffer(0));
```
so if you want the first half screen with scanlines repeated twice:
```
for (int i = 0; i < directCtrl.getViewPortHeight() / 2; i += 2) {
  directCtrl.setScanlineBuffer(i + 1, directCtrl.getScanlineBuffer(i));
}
```
Every call to your drawScanline() will draw two scanlines at the time. Of course only one when you are drawing the first half of the screen.

Could it be useful?

fdivitto, 2021-10-05:
I've just added the methods described above. Here is DirectVGA example modified to show how use them.
Notes:
 - drawScanline() will be called every two scanlines (look at setScanlinesPerCallBack())
 - drawScanline() will draw **one** scanline in the upper-half part of the screen (because they are doubled), while **two** scanlines in the lower-half (see the **for** loop)
 - you'll see the background is doubled in the upper-half, without the help of drawScanline(). Actually also the rows of red box are doubled, but of course we cannot see it

```
#include "fabgl.h"

fabgl::VGADirectController DisplayController;

volatile double      objX                 = 300;
volatile double      objY                 = 200;
static double        objDir               = 0.3;
static double        objVel               =   2;
static constexpr int objSize              =  25;
static uint32_t      bgParkMillerState    = 1;
static constexpr int borderSize           = 20;
constexpr int        scanlinesPerCallback = 2;  // screen height should be divisible by this value

static TaskHandle_t  mainTaskHandle;

// just to avoid floating point calculations inside MyDirectDrawVGAController::drawScanline()
volatile int objIntX;
volatile int objIntY;

inline int fastRandom()
{
  bgParkMillerState = (uint64_t)bgParkMillerState * 48271 % 0x7fffffff;
  return bgParkMillerState % 4;
}

void IRAM_ATTR drawScanline(void * arg, uint8_t * dest, int scanLine)
{
  auto fgcolor = DisplayController.createRawPixel(RGB222(3, 0, 0)); // red
  auto bgcolor = DisplayController.createRawPixel(RGB222(0, 0, 2)); // blue

  auto width  = DisplayController.getScreenWidth();
  auto height = DisplayController.getScreenHeight();

  // draws "scanlinesPerCallback" scanlines every time drawScanline() is called
  for (int i = 0; i < scanlinesPerCallback; ++i) {

    static bool bgcolorState = false;
    bgcolor = DisplayController.createRawPixel(bgcolorState ? RGB222(0, 0, 0) :  RGB222(2, 2, 2));
    bgcolorState = !bgcolorState;

    // fill line with background color
    memset(dest, bgcolor, width);

    // fill object with foreground color
    if (scanLine >= objIntY - objSize / 2 && scanLine <= objIntY + objSize / 2) {
      for (int col = objIntX - objSize / 2; col < objIntX + objSize / 2; ++col) {
        if (col >= 0 && col < width) {
          VGA_PIXELINROW(dest, col) = fgcolor;
        }
      }
    }

    // go to next scanline
    ++scanLine;
    dest += width;

    if (scanLine < height / 2)
      break;
  }

  if (scanLine == height) {
    // signal end of screen
    vTaskNotifyGiveFromISR(mainTaskHandle, NULL);
  }
}

void setup()
{
  mainTaskHandle = xTaskGetCurrentTaskHandle();

  DisplayController.begin();
  DisplayController.setScanlinesPerCallBack(scanlinesPerCallback);
  DisplayController.setDrawScanlineCallback(drawScanline);
  DisplayController.setResolution(VGA_640x480_60Hz);

  for (int i = 0; i < DisplayController.getViewPortHeight() / 2; i += 2)
    DisplayController.setScanlineBuffer(i + 1, DisplayController.getScanlineBuffer(i));
}

void loop()
{
  // test collision with borders and bounce changing direction
  if (objX < objSize / 2 || objX > DisplayController.getScreenWidth() - objSize / 2)
    objDir = PI - objDir;
  else if (objY < borderSize + objSize / 2 || objY > DisplayController.getScreenHeight() - borderSize - objSize / 2)
    objDir = 2 * PI - objDir;

  // calculate new coordinates
  objX += objVel * cos(objDir);
  objY += objVel * sin(objDir);

  // convert object coordinate to integer
  objIntX = objX;
  objIntY = objY;

  // wait for vertical sync
  ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
}
```

abelykh0, 2021-10-06:
Perfect, thank you!

closed 2021-10-06 by fdivitto