fdivitto/FabGL #120: [Feature] Adding own Menu-Items to PCEmulator Boot-Menu

guidol70, 2021-05-27:
Here my Feature-Request - as requested on YT ;)

Could you please add the ability to add own images to the boot-menu (or make the current also configureable)? In the past we had the chance to boot fd.img/hdd.img

So my first idea is to add only some generic lines to the boot-menu, which doenst try to download a image, but does it boot when its available on the SDcard.

I did this - for now - for my self - adding some lines for my MSDOS6.22 and own version of CPM-86.

But I think a editable config-file would be the best, because entering URLs via the onScreen-Menu wouldnt be that fast.

For some users it would also nice to disable WiFi (and NTP) for faster bootup.
Maybe there is also a chance to boot faster with WiFi when the board is only reseted and doesnt need to connect again (because it already connected? or is the connection on "reset" lost?)

And as the last wish - how about a static network setting (could this be faster) configured via onScreen menu or also the config-text-file?

fdivitto, 2021-08-22:
Feature partially implemented:
- now a configuration file is used (and created automatically if not present), where you can specify disks (url or an actual IMG file stored in SD card). So you can customize machines selection listbox.

Other options will be added in future (machine type, memory, graphic card, etc...etc...)

fdivitto, 2021-08-24:
- implemented a file selection dialog box, to select disk images stored in SD card
- now CTRL-ALT-DEL just resets the emulated PC, instead of ESP32
- pressing SYSREQ (ALT+PRINTSCREEN) resets the ESP32

fdivitto, 2021-09-30:
the latest commits should have made configurations more manageable

closed 2021-09-30 by fdivitto