fdivitto/FabGL #124: Bad charset for VK_CEDILLA_c ?

0xDRRB, 2021-06-09:
Hi. I am currently working on a French Keyboard layout.
keyboard.cpp seems to use UTF8/ISO-8859-1 charset as `VK_GRAVE_a` translate to 0xE0, `VK_ACUTE_e` to 0xE9, etc.

But `VK_CEDILLA_c` return 0x87 (CP437 encoding), not 0xE7 (UTF8/ISO-8859-1). So even with a good font the character is not the right one.

I don't think this deserves a PR on its own. I would submit one within a few days, just for the French layout.

GmEsoft, 2021-06-09:
I added the Belgian keyboard layout on my own fork, along with a fix for the dead keys handling using Alt Gr (Right Alt).
See https://github.com/GmEsoft/FabGL ... Feel free to merge my commit.

GmEsoft, 2021-06-09:
Also added in my fork some missing end-of-list markers at the end of the tables in KeyboardLayouts.

fdivitto, 2021-06-28:
I've just added French and Belgian layouts (thanks!).
Now conversion to ASCII is performed using tables (named codepages). Each font has of course its codepage, so if the character exists in the font it can be associated to a virtualkey (look at codepages.cpp).
Because I have only fonts belonging to cp437 and cp1252 codepages I have implemented just these. If you like to have other codepages just send me (with fonts)!

closed 2021-07-01 by fdivitto