fdivitto/FabGL #126: Request/Feature: VT100 Font for Terminal

guidol70, 2021-06-18:
Is there a chance to get a VT100 font like on the following
VT100 CRT Emulation?:
https://github.com/larsbrinkhoff/terminal-simulator

The emulations does run the VT100 Firmware in a 8080-Emulator.
So it sounds like a Z80-Emulator like on the VGA32 should also be a possible VT100?
Maybe only with a special font or like the complete emulation like the other github-project?

[image]

guidol70, 2021-07-14:
this one here could be more easy to convert. A [vt52 font in .otf](https://github.com/fritzm/vt52/blob/master/fonts/vt52.otf) format.
(I also converted it online to .ttf)
[image]
[vt52_ttf.zip](https://github.com/fdivitto/FabGL/files/6817030/vt52_ttf.zip)

fdivitto, 2021-10-14:
do you like to have a true VT100 font (meaning the character set) or just the CRT effect?
FabGL terminal needs a fixed width font, bitmapped. The common used format for this kind of font is "FON".

Using a real VT100 font means you will have a 8x9 font, that using 640x480 resolution results to 80x53 screen size.