fdivitto/FabGL #127: Show image (bmp, png, jpg or any) from SPIFF

e-labInnovations, 2021-06-20:
How to show an image (bmp, png, jpg or any) from SPIFF.

I would like to download some images file (library supported format from web server) to SPIFFS of ESP32 and show to the VGA monitor. 

### Problems I facing
- ESP32 crashes If I use
```
VGAController.begin(GPIO_NUM_12, GPIO_NUM_27, GPIO_NUM_14, GPIO_NUM_26, GPIO_NUM_25); //8 colours
```
Before
```
WiFiMulti.addAP(ssid, password);
while (WiFiMulti.run() != WL_CONNECTED) {
Serial.print(".");
delay(500);
}
```
- I don't know how to show a bmp or image file
- Can I show a 640x400 or similar sized image?

fdivitto, 2021-06-22:
at the moment there isn't an image file decoder, so you cannot do this using just fabgl. However writing a BMP decoder is a simple task: anyway consider that you have up to 64 colors, so you should implement also some sort of dithering algorithm to show photos, and the quality will be poor anyway...

e-labInnovations, 2021-07-01:
Hi @fdivitto 
How to show a bitmap something like this
```
(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t* bitmap) {
   //Bitmap from uint16_t* bitmap (RGB565)
  //canvas.drawBitmap function
}
```

I am trying to use [TJpg_Decoder](https://github.com/Bodmer/TJpg_Decoder) Library

e-labInnovations, 2021-07-02:
Working...
``` CPP
#include "fabgl.h"
#include <TJpg_Decoder.h>
#include <FS.h>
#include "SPIFFS.h" // ESP32 only

fabgl::VGA16Controller VGAController;
fabgl::Canvas cv(&VGAController);

// This next function will be called during decoding of the jpeg file to
// render each block to the Canvas.
bool vga_output(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t* bitmap) {
  // Stop further decoding as image is running off bottom of screen
  if ( y >= cv.getHeight() ) return 0;

  for (int16_t j = 0; j < h; j++, y++) {
    for (int16_t i = 0; i < w; i++) {
      writePixel(x + i, y, bitmap[j * w + i]);
    }
  }

  // Return 1 to decode next block
  return 1;
}

void setup() {
  Serial.begin(115200);
  Serial.println("\n\n Testing TJpg_Decoder library With FabGl");

  // Initialise SPIFFS
  if (!SPIFFS.begin()) {
    Serial.println("SPIFFS initialisation failed!");
    while (1) yield(); // Stay here twiddling thumbs waiting
  }
  Serial.println("\r\nInitialisation done.");

  VGAController.begin(GPIO_NUM_12, GPIO_NUM_27, GPIO_NUM_14, GPIO_NUM_26, GPIO_NUM_25);
  VGAController.setResolution(VGA_640x480_60Hz);

  cv.setBrushColor(RGB888(0, 0, 0));
  cv.clear();

  // The jpeg image can be scaled by a factor of 1, 2, 4, or 8
  TJpgDec.setJpgScale(1);

  // The decoder must be given the exact name of the rendering function above
  TJpgDec.setCallback(vga_output);

  // Time recorded for test purposes
  uint32_t t = millis();

  // Get the width and height in pixels of the jpeg if you wish
  uint16_t w = 0, h = 0;
  TJpgDec.getFsJpgSize(&w, &h, "/panda.jpg"); // Note name preceded with "/"
  Serial.print("Width = "); Serial.print(w); Serial.print(", height = "); Serial.println(h);

  // Draw the image, top left at 0,0
  TJpgDec.drawFsJpg(0, 0, "/panda.jpg");

  // How much time did rendering take
  t = millis() - t;
  Serial.print(t); Serial.println(" ms");

}

void loop() {
}

void writePixel(int16_t x, int16_t y, uint16_t rgb565) {
  uint8_t r = ((rgb565 >> 11) & 0x1F) * 255 / 31;   // red   0 .. 255
  uint8_t g = ((rgb565 >> 5) & 0x3F) * 255 / 63.0;  // green 0 .. 255
  uint8_t b = (rgb565 & 0x1F) * 255 / 31.0;         // blue  0 .. 255
  cv.setPixel(x, y, RGB888(r, g, b));
}
```

### Image I used
[image]

### Output on the monitor (8 colours only)

Miguel0101, 2021-07-12:
Does FabGL support USB keyboards?

closed 2021-10-04 by fdivitto