fdivitto/FabGL #130: serial input corruption..

WacKEDmaN, 2021-07-01:
im trying to hook up z80-mbc2 to vga32 via the gpio's through a bi-directional level shifter...
i started with both at 115200, but i see some text from the z80mbc2, but also alot of corruption (bad characters)
as soon as i add the wire to pin 2 it starts receiving input... 
i tried lowering the serial speed on both but still has garbled input on the vga32...

it seems fabgl is using software serial for this.. any chance to clean it up? or is there some fix i need?!

fdivitto, 2021-07-01:
no, fabgl (AnsiTerminal) doesn't use software serial: it uses hardware uart device.
Using other boards (ie https://www.tindie.com/products/fabgl/fabgl-terminal/) it just works, so I guess there is a problem with the lever shifter.

WacKEDmaN, 2021-07-01:
hmm ok.. i dont think its the level shifter coz its going crazy as soon as i connect the wire, with the other end free (like and antenna)
i have also tried another level shifter with the same results

fdivitto, 2021-07-01:
ok, but excluding a software problem, what else could it be??

WacKEDmaN, 2021-07-01:
not sure... usb serial works fine.. 
...like i said as soon as i connect a wire to the header the vga32 is seeing input... 
ive hooked up the usb-serial to it directly and its fine.. both 3.3v and 5v levels work perfectly fine.. so.. i did away with the level shifter and connected directly... with same garbled results :(

...i dont get it... i just hovered the wire above the pin about 2-3mm away... and it started receiving garbled input!... its acting like an antenna :/
...can i use either of the other gpios on the ttgo-vga32?

fdivitto, 2021-07-01:
Which ttgo-vga32 is? (V1.2, V1.4...)
Could you post a screenshot of what you see (bad chars)?
Finally, could you post the connection (board->level shifter->MBC) schema?

WacKEDmaN, 2021-07-01:
its V1.4
board is connected as follows...

IO34, IO2, 3v3 and gnd to 3.3v side of level shifter, 
TX, RX, 5v, gnd to 5v side of level shifter

heres a screenshot of just the single pin connected to IO2...other end of the wire flapping in the breeze!
https://imgur.com/Zh62FSZ

fdivitto, 2021-07-01:
On V1.4 the pin IO2 is used for SDCARD, but it should work without an sdcard inserted.

PS: of course I've ruled out all the obvious...bad baud rate, bits, parity....
PS2: could this help? [https://www.instructables.com/Using-ESP32VGA32-Terminal-for-Z80-MBC2/](https://www.instructables.com/Using-ESP32VGA32-Terminal-for-Z80-MBC2/)

WacKEDmaN, 2021-07-01:
no sdcard is inserted in the vga32...
i just wired up yet another level shifter, same results...
i looked at the instructables page earlier and thought better to use the level shifter, as i dont have them value resistors on hand...

its almost like the baud rates dont match, but they are fine with other serial devices :/
i will try the level shifting using 100k and 47k resistors...

WacKEDmaN, 2021-07-01:
same issue with resistors...

its really strange...the pin is VERY sensitive.. just touching it (with anything) causes garbled text or spaces/blocks...
if i connect the usb-serial to the vga32, it works perfectly fine at 115200bps...and WITHOUT the level shifting, eg, usb-serial set to 5v..

fdivitto, 2021-07-01:
I've tried to replicate....with the same your results!
It seems that 3v3 and GND pins of VGA32 are connected to something else.

Also, in my board it seems that IO14 is connected to GND, so you can get GND from it! 3V3 outputs 3.3V but...it doesn't allow the board to boot if you switch on MBC2 before VGA32, so probably it is connected to a strapping pin!

Just to recap:
- use IO14 as GND (connected to GND of level shifter) - >>> at your risk! :-)
- 3V3 to LV of level shifter
- other pins as usual

[image]
[image]
[image]
[image]

fdivitto, 2021-07-01:
I've just contacted TTGO asking for a check...

WacKEDmaN, 2021-07-01:
yes! that works great now...with IO14 as gnd... 
maybe the print on the board in wrong...

closed 2021-07-04 by fdivitto

WacKEDmaN, 2021-07-05:
sorry fab...but the corruption is back with IO14 as gnd... nothing changed! just started again out of nowhere :/
....removed the gnd all together.... and theres no/very very little corruption... (only have the gnd to the level shifter from the power supply and to the z80mbc2..no gnd for the RX/TX pins on the vga32)

seems IO14 is connected to something... removing the wire while it was running caused a reboot!

fdivitto, 2021-07-05:
So I guess you should get GND (and I suppose 3.3V also) somewhere else in the board. I'm sorry, of course I am not here to support VGA32 module, mine was just an attempt to help you!

I've added CTS/RTS to test my own terminal board, which compatible with both MBC2 (TTL UART) and with a classical RS232 host:
[image]
[image]

WacKEDmaN, 2021-07-05:
no probs mate... i guess you havent heard back from lilygo...
3v3 seems to be fine on this board... its just the gnd... but that said..ive been runnng it for the last hour with no gnd connection to the vga32... zero corruption!..

im starting to think there maybe a ground loop... as im powering the vga32 from the a seperate power supply to the z80mbc2.. the only gnd connected now is from the power supply to the level shifter and from the power supply to the vga32... guess i need to do some more testing.... ill drop my results here for anyone that may come across this issue...

fdivitto, 2021-07-05:
I've also a problem on the 3v3 pin: if I leave it connected to LV of level shifter the VGA32 doesn't boot up!

I'll add Reboot button on config dialog...

WacKEDmaN, 2021-07-05:
yeah thats weird...no issues with that here.. vga32 boots/reboots fine every time

WacKEDmaN, 2021-07-05:
after some more testing... with the level shifter connected to 3v3 of the vga32 board, it boots, but NVS fails!... it will save preferences, but it wont read them back!
```
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: TempResolution NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: Resolution NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: Columns NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: Rows NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: BGColor NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: Font NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: TermType NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: KbdLayout NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: BGColor NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: FGColor NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: SerCtl NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: BaudRate NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: Parity NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: DataLen NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: StopBits NOT_FOUND
[V][Preferences.cpp:377] getInt(): nvs_get_i32 fail: FlowCtrl NOT_FOUND
```

i removed 3v3... (only RX and TX connected to the vga32).. and seems to be much better.... no corruption (no gnd either!).. preferences load...
seems the level shifter isnt even needed....

WacKEDmaN, 2021-07-08:
got around to changing to single power supply for both boards... its rock solid now...SMH