fdivitto/FabGL #136: This app requires a SD-CARD!

eh, 2021-07-14:
Loading PCEmulator on a TTGO VGA32 v1.4 this error appears. It doesn't matter what size SD card, I have tried 128MB (yes Mega-Bytes) to 16GB. I can't find any info on what the SD card is supposed to contain so I've tried formatting it to fat16 and fat32. Nothing seems to work. I tried un-commenting:

// uncomment to format SD!
FileBrowser::format(fabgl::DriveType::SDCard, 0);

but this made no difference. Not sure what I'm missing. Any help?

guidol70, 2021-07-16:
You need a FAT(32)-formatted SDCard which should contain a bootable image
like
[FD.IMG](https://www.allbootdisks.com/downloads/Disks/MS-DOS_Boot_Disk_Download47/Diskette%20Images/Dos5.0.img) for a Floppy as A:
[HD.IMG](https://www.dropbox.com/s/e65d9749fi39qba/hd.img) for a HDD as C:

The .IMG format is the same as used by the DOSBOX-Project, so you may find a bootable Floppy-Image on the [net](https://www.allbootdisks.com/download/dos.html)

eh, 2021-07-16:
I've tried putting an image onto an sdcard (using dd) and also formatting the sdcard to fat32 and copying the .img file onto it. Either way I get "This app requires a SD-CARD!" on the screen.

eh, 2021-07-19:
Is it possible I have a strange V1.4 board with different pins to the SD card? Silkscreen has VGA32_V1.4 and a date of 21-2-20 below the SD card slot.

guidol70, 2021-07-19:
@eh Normally the pinout shouldnt differ. My board do have all the date 20-4-28 in the silkscreen.
It sounds as the VGA32 doenst recognize your MicroSD-Card. Did you try different ones?
Using dd for the image is the wrong way, you only have to put the .img files to the card.

Did you configure the WiFi? Because the last versions of the PC-Emulation does use WiFi to set the time via NTP and then download the boot-images via WiFi to the Card, but yout card isnt recognized.

Did you see the compile instructions in the source?
partition scheme must be: Huge App
make sure PSRAM is DISABLED

Search in the .INO of the PC-Emulation for http://www.fabglib.org/downloads/
there you could find the links for A: and C: images which are normally downloaded via WiFi

Maybe

eh, 2021-07-19:
> @eh Normally the pinout shouldnt differ. My board do have all the date 20-4-28 in the silkscreen.

Mine is 21-2-20. Is it possible to find out if the pinout has changed with this different revision? Please see https://github.com/fdivitto/FabGL/issues/74 where this is addressed previously I believe for V1.2 vs. V1.4.
Now we have 20-4-28 (yours) and 21-2-20 (mine) in the wild.

> It sounds as the VGA32 doenst recognize your MicroSD-Card. Did you try different ones?'

As I originally stated, yes, I tried many.

> Using dd for the image is the wrong way, you only have to put the .img files to the card.

I had also tried fat16 and fat32, copying the .img files to the card. The card is not detected.

> Did you configure the WiFi? Because the last versions of the PC-Emulation does use WiFi to set the time via NTP and then download the boot-images via WiFi to the Card, but yout card isnt recognized.

Yes I fully configured it and the result is the same, it won't download any images because the SD card isn't detected.

> Did you see the compile instructions in the source?
> partition scheme must be: Huge App
> make sure PSRAM is DISABLED

Yes, these are set correctly.

guidol70, 2021-07-20:
> Yes I fully configured it and the result is the same, it won't download any images because the SD card isn't detected.

Did you try to enable the format option for the SD-Card in Line 285-286 in the PCEmulation.ino?
```
  // uncomment to format SD!
  //FileBrowser::format(fabgl::DriveType::SDCard, 0);
```

The last change I did see for the SDCard Pin-Assignment was in [this commit](https://github.com/fdivitto/FabGL/commit/f9d09fcee69ab6b07e2cb82134f7817fab157a8f)

I also did try to ask for support via twitter - we will see ;)
[image]

eh, 2021-07-20:
> Did you try to enable the format option for the SD-Card in Line 285-286 in the PCEmulation.ino?

Yes, mentioned this in my original post.

> I also did try to ask for support via twitter - we will see ;)

Thanks! I tried passing several (but no-where near all) pin combinations to FileBrowser::mountSDCard but got nowhere. Hopefully they have some info.

guidol70, 2021-07-20:
From the commit I mentioned above - with the line
`return ChipPackage::ESP32PICOD4;  // TTGO-VGA32`
could you see when the sketch will be uploaded if your board is recognized in the flash-log as:
`Chip is ESP32-PICO-D4 (revision 1)`

eh, 2021-07-20:
I had a bit of a think about things and tried backing down the installed version of esp32 from 2.0.0-alpha1 to 1.0.6 and _TADA_ it works! It appears 1.0.6 is required, I just tried the recently released 2.0.0-rc1 and it completely fails to compile PCEmulator. Something about multiple WiFi libraries. I'm not going to fix that to find out more since we now have an answer.

Thanks for the help with this, frustrating issue. I appreciate it. Not sure if I should close this or not for future users that might run into this before a note can be made in the source.

guidol70, 2021-07-20:
>  It appears 1.0.6 is required

Yes "required" - but more supported. I had seen a "switch" [in the commits](https://github.com/fdivitto/FabGL/commit/602064f9602cabd3d69db63b280adeb4ad4db47c) when supporting a new ESP32 Support package in FabGL sources

And the last was the v1.0.6 ;)

smutek110, 2021-07-22:
You change the sd card pins in the FabGL \ src \ fabutils.h file inside the FabGL library

eh, 2021-07-22:
> You change the sd card pins in the FabGL \ src \ fabutils.h file inside the FabGL library

You can easily change them via FileBrowser::mountSDCard()

[http://www.fabglib.org/classfabgl_1_1_file_browser_a854da7be3c7cd8cc54d82171b978b52c.html](url)

smutek110, 2021-07-22:
for me, the effect was only after the change  in the FabGL \ src \ fabutils.h file inside the FabGL library. PCEmulator uses this configuration for SD Card.

eh, 2021-07-22:
PCEmulator uses mountSDCard() as seen here:

[PCEmulator.ino#L282](https://github.com/fdivitto/FabGL/blob/11f81e2bdd35bb7cd00af6b80a4eabc593216ac2/examples/VGA/PCEmulator/PCEmulator.ino#L282)

SD Card pins can be specified by changing as necessary, for example:

`if (!FileBrowser::mountSDCard(false, "/SD", 8, 16384, MISO, MOSI, CLK, CS ))`
`if (!FileBrowser::mountSDCard(false, "/SD", 8, 16384, 16, 17, 14, 13 ))`

smutek110, 2021-07-22:
I was able to launch the program, set up WIFI, download the image files and the program ends with a black screen

fdivitto, 2021-08-13:
it could be the VGA resolution is not supported by your monitor. Have you tried another one?

closed 2021-08-16 by fdivitto

coolcoder613eb, 2024-06-13:
I am having this issue too.

coolcoder613eb, 2024-06-16:
I resolved the issue.
I was using a slow 512MB SD card, when I tried a faster 16GB one, it worked.