fdivitto/FabGL #140: Display not appearing after image is run

i-am-shodan, 2021-08-10:
Hi I'm trying to get the PCEmulator sketch to work. I have a TTGO VGA32 1.4 and crucially a HDMI monitor.

I have altered the sketch to use ```ibox.begin(VGA_640x400_70Hz, 400, 300);```. I then get the set up menu on screen through my VGA converter. If I actually run an image then I lose VGA.

My guess was that the display mode is being set to CGA/EGA or something at a resolution beyond what my converter box can handle. I bought a [HD Video Converter Board,CGA/EGA/YUV/RGB to VGA Converter Board](https://www.amazon.co.uk/Converter-Arcade-Digital-Adapter-Monitor-default/dp/B08MTH58HR/ref=sr_1_7?dchild=1&keywords=cga+ega+converter&qid=1628596518&sr=8-7) in the hope that if I was running at low DOS resolution I would get something out of it.

I'm a bit stumped now so I have some questions:

- What is the resolution I should expect when ```machine->run();``` is called?
- Has anyone got HDMI conversion working?
- If the resolution has changed can I force it to something else with a custom DOS image?

i-am-shodan, 2021-08-10:
OK I understand some of the problem now. The converter I have is a GBS 8200 and converts analog RGB to VGA. The TTGO VGA32 output digital RGB.

I think that when the emulator boots up it moves into a super low res CGA graphics mode that the HDMI adapter I have does not support.

i-am-shodan, 2021-08-11:
Would it be possible to display a high res like VGA_640x480_60Hz but draw at a CGI resolution in the centre?

i-am-shodan, 2021-08-11:
I've had a go at hacking graphicsadapter.cpp and have been able to force 640x480 with this:
```
      case Emulation::PC_Text_40x25_16Colors:
        //printf("Emulation::PC_Text_40x25_16Colors\n");
        setFont(&FONT_8x16);
        setCursorShape(13, 15);
        m_VGADCtrl.setDrawScanlineCallback(drawScanline_PC_Text_80x25_16Colors, this);
        m_VGADCtrl.setScanlinesPerCallBack(8);
        m_VGADCtrl.setResolution(VGA_640x480_60Hz); // VGA_320x200_70Hz
        m_VGADCtrl.convertModelineToTimings(VGA_320x200_70Hz, &timings); // todo ret check

        viewPortWidth  = ~3 & timings.HVisibleArea; // view port width must be 32 bit aligned
        viewPortHeight = timings.VVisibleArea;

        m_columns = viewPortWidth / m_font.width;
        m_rows    = viewPortHeight / m_font.height;
        break;
```

FreeDos seems to work with some garbage at the bottom of the screen which I'm assuming is uninitialized memory.

i-am-shodan, 2021-08-12:
I've fixed text mode with this:
```
void IRAM_ATTR GraphicsAdapter::drawScanline_PC_Text_80x25_16Colors_Hack(void * arg, uint8_t * dest, int scanLine)
{
  constexpr int SCREENWIDTH      = 640;

  if (scanLine >= 400)
  {
    for (int x = 0; x < SCREENWIDTH * 8; ++x) {
        auto destptr = dest + x; 
        *destptr = 192; // black
    }
  }
  else
  {
    drawScanline_PC_Text_80x25_16Colors(arg, dest, scanLine);
  }
}
```

Can't get display mode to work with a similar hack. I'm guessing the colours or something need to be converted.

i-am-shodan, 2021-08-13:
I've now fixes all modes apart from PC_Graphics_HGC_720x348 to work with a HDMI adapter. I will send a PR.

fdivitto, 2021-08-13:
cutting a 480 resolution to 400 results in a ugly black band. Have you also tried to set "double scan"?

I can successfully use a VGA to HDMI converter using following changes:
VGA_320x200_70Hz -> VGA_320x200_60HzD      (used for 40x25 text mode)
VGA_640x400_70Hz -> VGA_640x400_60Hz        (used for 80x25 text mode)
VGA_320x200_70Hz -> VGA_320x200_60HzD      (used for CGA 320x200 mode)
VGA_640x200_70Hz -> VGA_640x200_60HzD      (used for CGA 640x200 mode)
VGA_720x348_73Hz -> VGA_720x348_50HzD       (used for Hercules high res mode)

Attached there is graphicsadapter.cpp with above changes.

[graphicsadapter.cpp.zip](https://github.com/fdivitto/FabGL/files/6984072/graphicsadapter.cpp.zip)

i-am-shodan, 2021-08-13:
I could find a list of compatible hdmi adapters anywhere so I bought two different ones on Amazon. Neither of the HDMI adapters I used supported VGA_640x400_60Hz. But I suppose it could also be the monitor I used.

Are any of these other modes documented anywhere? Is there a list of HDMI adapters?

fdivitto, 2021-08-13:
I use the cheapest VGA2HDMI converter there is around (this one shorturl.at/atRW6)...

i-am-shodan, 2021-08-13:
Your link 404s

fdivitto, 2021-08-13:
link just updated

fdivitto, 2021-08-13:
my converter supports 640x400 at 60Hz.
However this night I will try to find a double scan resolution which is actually a 640x480 (but you will be 640x240, so we will try to remove the ugly black band)...

i-am-shodan, 2021-08-13:
I wouldn’t rush, I’ve literally spent 3 days authoring a PR to fix this.

fdivitto, 2021-09-27:
not the best solution, but all resolutions have been now adjusted to be usable on most monitors and adapters

closed 2021-09-27 by fdivitto