fdivitto/FabGL #143: PCEmulator: not proper display

Unixware, 2021-08-16:
I am using 3 VGA monitors (2x 4:3, all of them TFTs)  and the display is quite problematic, on one monitor the top and bottom lines are a bit "stretched" (not visible) -even with auto-adjust there is no way to contain them into the visible display.
Only in one VGA monitor is , even then the bottom line is half-visible... and the power button LED is blinking -that means the display signals are reaching the limits.  (tests with the today's  (16/8/21) version of github)
 thanks

fdivitto, 2021-08-16:
I know, see also [this](https://github.com/fdivitto/FabGL/issues/140)

I hope to find a solution soon...

Unixware, 2021-08-16:
yes, I already used that file, top line is ok but bottom is not visible at all, many thanks!

fdivitto, 2021-08-16:
we need to find some modelines to display following resolutions (used in PC Emulator) which work on most VGA monitors:

CGA low: 320x200 (now VGA_320x200_70Hz, alternative VGA_320x200_60HzD)
CGA high: 640x200 (now VGA_640x400_70Hz, alternative VGA_640x400_60Hz)
Hercules: 720x348 (now VGA_720x348_73Hz, alternative VGA_720x348_50HzD)

"ModelineStudio" application is useful to experiment with modelines.

Unixware, 2021-08-16:
thanks, no worries. beating a dead horse here...

closed 2021-08-16 by Unixware