fdivitto/FabGL #144: Incorrect colors in CGA mode

i-am-shodan, 2021-08-17:
I created a custom boot disk for Defender of the Crown. It loads but the colours are wrong.

- Defender of the Crown uses low intensity pallet 0 but high intensity is in use
- The background color is wrong, white is being used instead of black

I think I can repo this with pac-man.exe. The game is meant to look like this [playable game here](https://www.retrogames.cz/play_017-DOS.php) but actually looks like garbage.

I added some debugging to the setPCGraphicsPaletteInUse and setPCGraphicsBackgroundColorIndex calls for defender of the crown and this is the output:
```
Emulation::setPCGraphicsPaletteInUse 0
Emulation::setPCGraphicsBackgroundColorIndex 0
Emulation::setPCGraphicsPaletteInUse 3
Emulation::setPCGraphicsBackgroundColorIndex 0
Emulation::setPCGraphicsPaletteInUse 1
Emulation::setPCGraphicsBackgroundColorIndex 15
Emulation::setPCGraphicsPaletteInUse 3
Emulation::setPCGraphicsBackgroundColorIndex 0
Emulation::setPCGraphicsPaletteInUse 1
Emulation::setPCGraphicsBackgroundColorIndex 15
Emulation::setPCGraphicsPaletteInUse 3
Emulation::setPCGraphicsBackgroundColorIndex 0
Emulation::setPCGraphicsPaletteInUse 1
Emulation::setPCGraphicsBackgroundColorIndex 15
Emulation::setPCGraphicsPaletteInUse 3
Emulation::setPCGraphicsBackgroundColorIndex 0
Emulation::setPCGraphicsPaletteInUse 1
Emulation::setPCGraphicsBackgroundColorIndex 15
Emulation::setPCGraphicsPaletteInUse 3
Emulation::setPCGraphicsBackgroundColorIndex 0
Emulation::setPCGraphicsPaletteInUse 1
Emulation::setPCGraphicsBackgroundColorIndex 15
Emulation::setPCGraphicsPaletteInUse 3
Emulation::setPCGraphicsBackgroundColorIndex 0
Emulation::setPCGraphicsPaletteInUse 1
Emulation::setPCGraphicsBackgroundColorIndex 15
Emulation::setPCGraphicsPaletteInUse 3
Emulation::setPCGraphicsBackgroundColorIndex 0
Emulation::setPCGraphicsPaletteInUse 1
Emulation::setPCGraphicsBackgroundColorIndex 15
```

i-am-shodan, 2021-08-17:
If I change the setPCGraphicsBackgroundColorIndex function to ignore the background color change to 15 (white) then I get something that is at least playable. Graphics are in the high intensity colour range but otherwise OK.
My guess is that changing to low intensity colours isn't working / maybe the CGA mask bits in  machine.cpp are wrong.

```
void GraphicsAdapter::setPCGraphicsBackgroundColorIndex(int colorIndex)
{
  printf("Emulation::setPCGraphicsBackgroundColorIndex %d\n", colorIndex);

  if (colorIndex == 15)
  {
    printf("Emulation::forcing colorIndex to 0\n");
    colorIndex = 0;
  }

  m_PCGraphicsBackgroundColorIndex = colorIndex;

  setupLUT();
}
```

fdivitto, 2021-09-27:
thank you, the last commit fixed this!

closed 2021-09-27 by fdivitto