fdivitto/FabGL #152: [Issue] with ESP32 2.0.0rc1 commit

guidol70, 2021-09-11:
Hi,
when using the fabutils.cpp from the commit
https://github.com/fdivitto/FabGL/commit/cf980b404751df290c30dbe5d7505afff684a3ba

I do get an error like
"File - not declared in scope" when compiling RunCPM for VGA32 with FabGL :(
I did also try to install the final ESP32 2.0.0 (not RC1) - but do get the same error.

When using the newest FabGL-master (25.08.2021), but inserting the fabutils.cpp from the 02.08.2021 I could compile with my ESP32 HW-Support v1.0.6

Is this new fabutils.cpp only for ESP 2.0.0rc1 or should it work with 2.0.0 non-RC1?

fdivitto, 2021-09-12:
Hi,
have you tried to compile an embedded example (like HardwareTest)?
I have just tested  2.0.0rc1, 2.0.0rc2 and of course the last release 1.0.6, and fabgl compiles correctly. I cannot find a "final" 2.0.0, just Release Candidates 1 and 2.

guidol70, 2021-09-12:
@fdivitto I dont think its original related to the FabGL-Code - only how this RunCPM-version does use FabGL and File-functions.
After the new fabutils.cpp the follwing code-part doenst seem to work (not declared in scope):
```
/* Memory abstraction functions */
/*===============================================================================*/
bool _RamLoad(char* filename, uint16 address) {
	File f;
	bool result = false;

	if (f = SD.open(filename, FILE_READ)) {
		while (f.available())
			_RamWrite(address++, f.read());
		f.close();
		result = true;
	}
	return(result);
}

```

I thought the final ESP32 Core 2.0.0 is available via the follwoing .json adress, because it isnt marked after install as rc1 nor rc2:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
Got this adress from
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

fdivitto, 2021-09-12:
ok, now I have 2.0.0. Espressif in "dev" repo reported up to 2.0.0rc2. Anyway even with 2.0.0 fabgl compiles correctly.
Please remember me where I can download a "all-in-one" zip with RunCPM.

guidol70, 2021-09-12:
> Please remember me where I can download a "all-in-one" zip with RunCPM.

[here](https://github.com/guidol70/RunCPM_VGA32/raw/main/v5_5/RunCPM_v5_5_VGA32_20210801.zip) you can grab it
on [my github](https://github.com/guidol70/RunCPM_VGA32)

fdivitto, 2021-09-14:
I'm experiencing a lot of issues with 2.0.0. Espressif changed something in SD card handling. "2.0.0" allocates much more RAM than before (using WiFi, SD...) so some applications just do not run or cannot mount SD card.

I guess your RunCPM port problem may be related to 2.0.0 instead of FabGL. It doesn't find "File" type, which is not a FabGL class, but and Arduino one.

Updates follow...

fdivitto, 2021-09-14:
last commit fixed some SD Card issues with core 2.0.0. 

It doesn't fix the unknown "File" type of RunCPM.

guidol70, 2021-09-14:
For me it looks like a problem in the sequence of including files/librarys in the gobal section of the -ino-sourcecode - which I cant solve by myself :(

Another "change/commit which I may find "important" is:
[Rename LITTLEFS to LittleFS to match ESP8266](https://github.com/espressif/arduino-esp32/commit/c8a4010f) in ESP32 Core 2.0.0

fdivitto, 2021-09-14:
After several hours lost I guess there are too many problems with 2.0.0, so at the moment I prefer to stay with 1.0.6: no more 2.0.0 issues will be handled.

I prefer to spend the little time I have adding new features rather than keeping up with the 2.0.0!

https://github.com/espressif/arduino-esp32/issues/5630
https://github.com/espressif/arduino-esp32/issues/5474
and so on...

guidol70, 2021-09-15:
> After several hours lost I guess there are too many problems with 2.0.0, so at the moment I prefer to stay with 1.0.6: no more 2.0.0 issues will be handled.
> 
> I prefer to spend the little time I have adding new features rather than keeping up with the 2.0.0!

I think you are right - its much to early to use 2.0.0 with these memory-leaks/footprints.
For RunCPM VGA32 I will also stay - for now - with 1.0.6.

I will look forward for your new great fetaures :)

guidol70, 2021-09-21:
> last commit fixed some SD Card issues with core 2.0.0.
> 
> It doesn't fix the unknown "File" type of RunCPM.

for "File not declared in scope" I found an idea - but didnt help also.
Its 
`#define FS_NO_GLOBALS`
before the includes, because it seems that **SPI.h/SDFat.h** also uses **File** 
like the in the **vfs_api.h** included **FS.h (FS 1.0)**, but in the **FS.h** there is at the end an entry for
```
#ifndef FS_NO_GLOBALS
using fs::FS;
using fs::File;
using fs::SeekMode;
using fs::SeekSet;
using fs::SeekCur;
using fs::SeekEnd;
#endif //FS_NO_GLOBALS
```

You didnt use the SDFat.h for accessing the SDCard - in the VIC20.ino you seem only to use stdio.h and the fabgl.h

Compile which does work:
```
Using library SPI at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SPI 
Using library SdFat at version 2.1.1-beta.1 in folder: C:\Users\guido\Documents\Arduino\libraries\SdFat 
Using library FabGL at version 1.0.4 in folder: C:\Users\guido\Documents\Arduino\libraries\FabGL 
Using library Preferences at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Preferences 
Using library WiFi at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi 
```

Compile which does fail ( **Using library FS at version 1.0** ) :
```
Using library SPI at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SPI 
Using library SdFat at version 2.1.1-beta.1 in folder: C:\Users\guido\Documents\Arduino\libraries\SdFat 
Using library FabGL at version 1.0.4 in folder: C:\Users\guido\Documents\Arduino\libraries\FabGL 
Using library Preferences at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Preferences 
Using library FS at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS 
Using library WiFi at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi 
exit status 1
'File' was not declared in this scope
```

guidol70, 2021-09-21:
a little **HEUREKA** ;)
Now I can compile with the latest **fabutils.cpp**
(using **fabutils.cpp** -> **vfs_api.h** -> **FS.h**)
It was a **problem** between **FS.h** and the **SDFat.h** which I do use for RunCPM for VGA32.

While using the ESP32 Core **v1.0.6** it does **compile & run**,
but when using ESP32 Core **v2.0.0** it does **compile and then hanging in a reboot-loop** while displaying the following **message at the serial-monitor**:
```
Rebooting...
assertion "xTaskToNotify" failed: file "IDF/components/freertos/tasks.c", line 5308, function: xTaskGenericNotify
abort() was called at PC 0x4010e18b on core 0
Backtrace:0x40104ac1:0x3ffb26600x4008aa3d:0x3ffb2680 0x4009063d:0x3ffb26a0 0x4010e18b:0x3ffb2720 0x4008d8c1:0x3ffb2750 0x400f31b9:0x3ffb2770 0x400f32af:0x3ffb27a0 0x400e05c1:0x3ffb27c0 0x400f92e3:0x3ffb2820 
ELF file SHA256: 0000000000000000
```
I also wrote a little bit more information/solution about this RunCPM for VGA-issue at
[[Info] compile Problem in RunCPM for VGA32 between SDFat.h and FS.h](https://github.com/MockbaTheBorg/RunCPM/issues/134)
while the **reboot-issue with v2.0.0 isnt resolved :(**

fdivitto, 2021-09-21:
there is an nice tool for Arduino/ESP32, which pasting the "Backtrace" row allows you to detect the line and callstack of the exception:

https://github.com/me-no-dev/EspExceptionDecoder

guidol70, 2021-09-21:
> there is an nice tool for Arduino/ESP32, which pasting the "Backtrace" row allows you to detect the line and callstack of the exception:

but isnt that not already included with the following text in the error-message?:
```
failed: file "IDF/components/freertos/tasks.c" 
line 5308
function: xTaskGenericNotify
```
which doesnt help me much :(

fdivitto, 2021-09-21:
it is useful to know the "full" callstack, so you can see the origin (for example the code in RunCPM or FabGL which eventually caused the exception).

guidol70, 2021-09-21:
> it is useful to know the "full" callstack, so you can see the origin (for example the code in RunCPM or FabGL which eventually caused the exception).

Is it possible, that the v2.0.0 of the ESP32 Core doesnt like making Sound via FabGL?:

```
Decoding stack results
0x40104ac1: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4008aa3d: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 129
0x4009063d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c line 46
0x4010e18b: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c line 62
0x4008d8c1: xTaskGenericNotify at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 5308
0x400f31b9: fabgl::SoundGenerator::forcePlay(bool) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\devdrivers\soundgen.cpp line 557
0x400f32af: fabgl::SoundGenerator::play(bool) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\devdrivers\soundgen.cpp line 525
0x400e05c1: setup() at C:\Users\guido\Documents\Arduino\libraries\FabGL\src/devdrivers/soundgen.h line 142
0x400f92e3: loopTask(void*) at C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32\main.cpp line 38
```

Would it help to eliminate the sound-commands I use for Keyboard-Click / Boot- and Reboot-Sound?

fdivitto, 2021-09-21:
ok, let me do some tests...

guidol70, 2021-09-22:
The RunCPM for VGA32 compiled with v2.0.0 and no Boot- & KeyClickSound cant use keyboard, but **can be controlled via SERCTL = USB-Serial-Port/-Terminal**.
Compile Backtrace was from the "Boot-Sound": 
```
// =========================================================================================
// Boot-Sound
// =========================================================================================
   for (int i = 0; i <= 2; i++)   {
       Terminal.soundGenerator()->playSound(SquareWaveformGenerator(), 800 , 50);
       Terminal.soundGenerator()->playSound(SquareWaveformGenerator(), 1600 , 50);
       delay(100);
                                  }
```
When only the Boot-Sound is deactivated the sketch did compile without Backtrace

Your test sketches **KeyboardScanCodes** and **KeyboardVirtualKeys** do run when compiled here with v2.0.0 and the same keyboard which wont accept any input with RunCPM.

guidol70, 2021-09-22:
> ok, let me do some tests...

@fdivitto [here is the "Test-Version"](https://github.com/guidol70/RunCPM_VGA32/raw/main/v5_6/RunCPM_v5_6_VGA32__File32_NoBootSound_20210921.zip) of RunCPM for VGA32 with **File32** for SDFat
(**because of vfs_api.h**) and the **disabled Boot-Sound** - so it can compile with v2.0.0 
but **keyboard input doesnt work** :(

This may be more easy for you to test this ;)

fdivitto, 2021-09-23:
> (**because of vfs_api.h**) and the **disabled Boot-Sound** - so it can compile with v2.0.0
> but **keyboard input doesnt work** :(
> 
> This may be more easy for you to test this ;)

I'm getting "Unable to initialize SD card" and next "Bdos Err on A: Select", and stop!
Of course I've put "A" content into "/A/0"...

It is clear that it cannot mount the SD card. 
However, in my tests (i.e. "HardwareTest" application) current snapshot of FabGL with core 2.0.0, can use Sounds, SD Card and WiFi, all together.

Any suggestion?

guidol70, 2021-09-23:
> I'm getting "Unable to initialize SD card" and next "Bdos Err on A: Select", and stop!
> Of course I've put "A" content into "/A/0"...
> It is clear that it cannot mount the SD card.
> Any suggestion?
On my VGA32 v1.4 (you did change the Hardware-Line or did you also use a VGA32 v1.4?)
it can init the card, because via USB-serial I can do a DIR and can start WordStar, but not from the PS/2-Keyboard.
Did you try to lower the Mhz for the SDCard?

I will do a test with the "HardwareTest" appilcation.
Can you please take a look if the lines in the first tab of the .ino has "actual" supported kexboard-code-lines (the VK-ones)? Thanks in advance.

fdivitto, 2021-09-23:
> Can you please take a look if the lines in the first tab of the .ino has "actual" supported kexboard-code-lines (the VK-ones)? Thanks in advance.

Do you mean if onVirtualKeyItem generates correct VK_ codes?

guidol70, 2021-09-23:
Reset Backtrace was (seems to) due my following (old) lines for the buttons
(now commented out):
```
// pinMode  (RESET_PIN, INPUT);
// pinMode(PREFCLR_PIN, INPUT);
// attachInterrupt(digitalPinToInterrupt(RESET_PIN), _esp_reboot, FALLING);
// attachInterrupt(digitalPinToInterrupt(PREFCLR_PIN), _pref_clr, FALLING);
```

guidol70, 2021-10-02:
Today I did try to rearrange my commands in the RunCPM .ino equal to the sequence to ANSITerminal or LoopbackTerminal to find "my error".

I came to the strange thing that I do get a Backtrace if I switch the sequence between
Terminal and ConfDialogApp :(

**Sequence which is working**
```
  Terminal.begin(&DisplayController);
  Terminal.clear();
  Terminal.enableCursor(true);
  ConfDialogApp::loadConfiguration();
```
**Sequence which gives me a Backtrace**
```
  ConfDialogApp::loadConfiguration();
  Terminal.begin(&DisplayController);
  Terminal.clear();
  Terminal.enableCursor(true);
```
**The Backtrace**
```
Decoding stack results
0x400875c0: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008783d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x400881ea: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 719
0x400e74a2: fabgl::Terminal::addToInputQueue(unsigned char, bool) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\terminal.cpp line 1907
0x400e7962: fabgl::Terminal::write(unsigned char, bool) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\terminal.cpp line 1923
0x400e7980: fabgl::Terminal::write(unsigned char) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\terminal.cpp line 1945
0x40119012: fabgl::TerminalController::write(unsigned char) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\terminal.cpp line 4972
0x400e9170: fabgl::TerminalController::setTerminalType(fabgl::TermType) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\terminal.cpp line 4981
0x400eba8d: fabgl::Terminal::setTerminalType(fabgl::TermType) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\terminal.cpp line 1961
0x400d203a: ConfDialogApp::loadConfiguration() at Q:\1A_RunCPM\RunCPM_v5_7_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_7_File32_3nd_v1_0_6_02102021/confdialog.h line 278
0x400db5f3: setup() at Q:\1A_RunCPM\RunCPM_v5_7_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_7_File32_3nd_v1_0_6_02102021/RunCPM_VGA32_v5_7_File32_3nd_v1_0_6_02102021.ino line 129
0x400f3b66: loopTask(void*) at C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 18
0x400888c6: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
So I came up finding **Terminal.connectLocally();**  in my **ConfDialogApp::loadConfiguration();** lines.

I did try to add this command additionally to the Terminal.-Lines but it isnt better.

But I found interesting, that when compiling RunCPM with ESP32 Core v1.0.6 and **not using** Terminal.connectLocally(); I have the **same non-input behaviour** like when compiled with v2.0.0 **and using** the line **Terminal.connectLocally();**

But it doesnt seems to be a problem directly with FabGL, because the LoopbackTermial
uses also **Terminal.connectLocally();** and does work.
Additionally the LoopbackTerminal also can use my Boot-Sound with v2.0.0 which gives me a Backtrace when I do compile RunCPM with v2.0.0 :(
```
// ==========================================================
// Boot-Sound
// ==========================================================
     for (int i = 0; i <= 2; i++)   {
         Terminal.soundGenerator()->playSound(SquareWaveformGenerator(), 800 , 50);
         Terminal.soundGenerator()->playSound(SquareWaveformGenerator(), 1600 , 50);
         delay(100);
                                    }

```

fdivitto, 2021-10-03:
it seems that FreeRTOS function xQueueGenericSend fails. It could be there is no enough memory to allocate the queue.
Just a try, please reduce the input queue size, for example to 512, so in your ".ino", instead of:

```
  // 2048 good to pass vttest
  Terminal.inputQueueSize = 2048; 
```

set 

```
Terminal.inputQueueSize = 512; 
```

BTW: are you checking (internal) free memory available?

guidol70, 2021-10-03:
> set
> ```
> Terminal.inputQueueSize = 512; 
> ```
> BTW: are you checking (internal) free memory available?

I dont know how to check the (internal) free memory :(
But you hint with the **Terminal.inputQueueSize** led me to another little memory-eater -
the **Terminal.keyboardReaderTaskStackSize** which is set to 3000

So I did set 
```
  // more stack is required for the UI (used inside Terminal.onVirtualKey)
  // default value 2048
  Terminal.keyboardReaderTaskStackSize = 2048;
  // Terminal.keyboardReaderTaskStackSize = 3000;
  
  // 2048 good to pass vttest
  // 1024 is default value
  //  512 good to save memory
  // Terminal.inputQueueSize = 2048;
  Terminal.inputQueueSize = 512;

```
With these settings I could input **one** char and then I do get a backtrace:
```
Decoding stack results
0x4010230d: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4008aa3d: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 129
0x4009063d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c line 46
0x4010b9d7: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c line 62
0x4008d8c1: xTaskGenericNotify at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 5308
0x400f0d41: fabgl::SoundGenerator::forcePlay(bool) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\devdrivers\soundgen.cpp line 557
0x400f0e37: fabgl::SoundGenerator::play(bool) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\devdrivers\soundgen.cpp line 525
0x400d2389: fabgl::Delegate :: ::_FUN(void *, fabgl::VirtualKey * const&, const bool &) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src/devdrivers/soundgen.h line 142
0x400ee676: fabgl::Terminal::keyboardReaderTask(void*) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\terminal.cpp line 4611
```
which does point to my KeyClick-Sound.
So I **disabled the KeyClick** like the Boot-Sound.

After that I **could do a normal input via the keyboard** :)
So the main RunCPM does start (and work)
**BUT** without the **Terminal.keyboardReaderTaskStackSize = 3000;**
it seems that I cant start the (F12)-ConfigDialog anymore, because then I do get the following Backtrace:
```
Decoding stack results
0x4008e26f: vPortEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h line 109
0x400853da: esp_crosscore_int_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/crosscore_int.c line 119
0x40085429: esp_crosscore_int_send_yield at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/crosscore_int.c line 144
0x4008b991: xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32/include/hal/cpu_ll.h line 39
0x4008ba1c: xQueueTakeMutexRecursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 687
0x40086a85: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 153
0x40086c29: __retarget_lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 323
0x40113ab9: _vfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c line 853
0x4010c04d: fiprintf at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fiprintf.c line 48
0x4010bfd5: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c line 58
0x4008b262: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 770
0x400e2872: fabgl::uiApp::postEvent(fabgl::uiEvent const*) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\fabui.cpp line 564
0x400f0675: fabgl::Keyboard::postVirtualKeyItem(fabgl::VirtualKeyItem const&) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp line 581
0x400f0716: fabgl::Keyboard::SCodeToVKConverterTask(void*) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp line 671
```
So only a partly success ;)
It seems really a problem with the higer memory-usage of ESP32 Core v2.0.0

Do you got a hint where to save memory on FabGL or Arduino-Core (WiFi isnt used) parts?

BTW: The same sketch does also compile fine with ESP32 Core v1.0.6
with lower memory demand (also could use the audio parts and the ConfigDialog)

guidol70, 2021-10-03:
@fdivitto 
When compiling RunCPM for VGA32 - while compiling FabGL - 
there are some entrys which I think maybe not needed for Z80/RunCPM
while Iam using the VGA16Controller
(other Graphics-Driver or Chips for VIC20 (6502) or PC-Emulation):
```
Compiling library "FabGL"
tsi2c.cpp.o
DS3231.cpp.o
MCP23S17.cpp.o
soundgen.cpp.o
TFTControllerGeneric.cpp.o
TFTControllerSpecif.cpp.o
SSD1306Controller.cpp.o
vga2controller.cpp.o
vga4controller.cpp.o
vga8controller.cpp.o
MOS6502.cpp.o
PIC8259.cpp.o
PIT8253.cpp.o
MC146818.cpp.o
VIA6522.cpp.o
i8042.cpp.o
i8086.cpp.o
```
Do they take up space which can be used for the Keyboard or Sound when not included - or arent they in the .bin after compilation?

fdivitto, 2021-10-03:
> Do they take up space which can be used for the Keyboard or Sound when not included - or arent they in the .bin after compilation?

no, the linker get rid of them.
Anyway we have to search for used internal RAM, not Flash or PSRAM.
You can know how much internal RAM is available with:
```
printf("Free Memory: %d bytes\r\n", heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL));
```

guidol70, 2021-10-03:
> Anyway we have to search for used internal RAM, not Flash or PSRAM. 
> 
Yes - with v2.0.0 there is very few RAM left.
**RunCPM VGA32 ESP32 Core v2.0.0 without Sound** 
```
Free Memory: 3383 Bytes
Free DMA Memory    : 3383
Free 32 bit Memory : 58435
```
with v1.0.6 we have a "few" bytes more of RAM for disposition
**RunCPM VGA32 ESP32 Core v1.0.6 with Sound** 
```
Free Memory: 13772 Bytes
Free DMA Memory    : 13772
Free 32 bit Memory : 85424
```
**Code used** 
```
Terminal.printf("Free Memory: %d bytes\r\n", heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL));
Terminal.printf("\e[32mFree DMA Memory    :\e[33m %d\r\n", heap_caps_get_free_size(MALLOC_CAP_DMA));
Terminal.printf("\e[32mFree 32 bit Memory :\e[33m %d\r\n\n", heap_caps_get_free_size(MALLOC_CAP_32BIT));
```

guidol70, 2021-10-05:
OK - now with VGA8Controller for ESP32 Core v2.0.0 all seems to be working fine :)
I have rearranged the font color to fit the design more to the available 8 colors.

I think we can close now this issue ;)

Here the "final" Boot-Screen-Result
[image]
:

closed 2021-10-05 by guidol70

SuGlider, 2021-10-14:
@guidol70 

In order to install it as a separated board, in IDE board manager, for testing, please follow the instructions from
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html
The only difference here would be to clone mem-optimized branch instead.

git clone -b mem-optimized https://github.com/espressif/arduino-esp32 esp32

Or just go to the URL and download the ZIP file with the repository.

Don't forget to run tools/get.exe ou get.py for Linux.