fdivitto/FabGL #168: Backtrace with ConfigDialog with FabGL-master from 09.10.2021

guidol70, 2021-10-09:
These days I had a working RunCPM with the FabGL-master from the 02.10.2021.
ConfDialog did work with VGA16Controller (ESP 1.06) 
and VGA8Controller ( ESP 1.06 & 2.0.0)

Today I did try the **FabGL-master from the 09.10.2021** and had a slow drawing ConfDialog with VGA8Controller and a **Backtrace while starting ConfDialog with VGA16Controller**.

I know you are hardworking - the last weeks - on new features for the Menue-possibilities (and the PC Emulator).

Maybe the Backtrace is only (again) due to low(er) memory because the new Menu (FabUI?) features in my RunCPM.

Can you please have a short look at the backtrace?
While using the FabGL-master from the 02.10.2021 all is back to working :)

The actual Backtrace:
```
Decoding stack results
0x400875c0: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008783d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x40115c2f: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x40115c76: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x4011535f: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x40115692: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400d3d81: ConfDialogApp::init() at Q:\1A_RunCPM\RunCPM_v5_7_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_7_BootScreen_Design_09102021/confdialog.h line 169
0x400df991: fabgl::uiApp::processEvent(fabgl::uiEvent*) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\fabui.cpp line 648
0x400e4c3d: fabgl::uiApp::run(fabgl::BitmappedDisplayController*, fabgl::Keyboard*, fabgl::Mouse*) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src\fabui.cpp line 307
0x400dba7a: fabgl::Delegate :: ::_FUN(void *, fabgl::VirtualKeyItem * const&) at Q:\1A_RunCPM\RunCPM_v5_7_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_7_BootScreen_Design_09102021/RunCPM_VGA32_v5_7_BootScreen_Design_09102021.ino line 377
0x400e93ea: fabgl::Terminal::keyboardReaderTask(void*) at C:\Users\guido\Documents\Arduino\libraries\FabGL\src/fabutils.h line 411
0x400888c6: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

fdivitto, 2021-10-09:
what do you have at line 169 of confdialog.h?

guidol70, 2021-10-09:
> what do you have at line 169 of confdialog.h?
> BTW: the configuration dialog of AnsiTerminal seems to work.

its only Exit without save at line 169
[image]

fdivitto, 2021-10-09:
Found the problem. I've added some delegates to uiWindow, which get few bytes per control. 
This can happen as RUNCPM is working at the limit of available RAM, so a little change like it can cause this kind of problems.

I've removed that delegates and now the dialog opens correctly on 1.0.6 and VGA16Controller.
In the night (or tomorrow...) I'll release a commit to try to fix this.

guidol70, 2021-10-10:
Before the new memory-Report 
**another problem with the FabUI from FabGL-master of the 09.10.2021**:

When Iam in the Config-Dialog I have to type 2x Tab for switching from one list-entry to another (so from Terminal-Type to Keyboard-Layout), but for Click-Entrys (like KeyClick and BootInfo) I need onyl to type one Tab :(

With the first Tab the (blue) ConfigDialog-Windows looses its focus (gets grey) and on the second Tab its gets the focus back and jumps to the next entry.

With FabGL-master from the 02.10.2021 I didnt have to double-tab for the list entrys.

Here the new Memory-report with some settings-changes
(**should I use Terminal.keyboardReaderTaskStackSize with under 3000?**
because also the ANSI-terminal does use the value of 3000)

> * ESP32 core **1.0.6**
>   * with current snapshot

and these settings:

```
  // more stack (2048 -> 3000) is required for the UI (used inside Terminal.onVirtualKey)
  // default value 2048
  Terminal.keyboardReaderTaskStackSize = 2048;
  // Terminal.keyboardReaderTaskStackSize = 3000;
  
  // 2048 good to pass vttest
  // 1024 is default value
  //  512 good to save memory
  // Terminal.inputQueueSize = 2048;
  Terminal.inputQueueSize = 512;

  // 1024 is default value
  //  512 is that good to save memory?
  DisplayController.queueSize = 512;

```

**VGA16Controller:** 
```
FabGL-master 09.10.2021
DEBUG: Free DMA Memory 8388 bytes
DEBUG: Free Memory 79836 bytes
```
```
FabGL-master 02.10.2021
DEBUG: Free DMA Memory 9736 bytes
DEBUG: Free Memory 81184 bytes
```

guidol70, 2021-10-10:
ESP32 Core v1.0.6
actual Snapshot
FabGL-Master **02.10.2021**

**VGAController8:**

Terminal.keyboardReaderTaskStackSize = 2048;

```
DEBUG: Free DMA Memory 48120 bytes
DEBUG: Free Memory     119524 bytes
```

Terminal.keyboardReaderTaskStackSize = 3000;

```
DEBUG: Free DMA Memory 47180 bytes
DEBUG: Free Memory     118584 bytes
```

guidol70, 2021-10-10:
> Before the new memory-Report **another problem with the FabUI from FabGL-master of the 09.10.2021**:
> 
> When Iam in the Config-Dialog (RunCPM **or also ANSITerminal**) I have to type 2x Tab for switching from one list-entry to another (so from Terminal-Type to Keyboard-Layout), but for Click-Entrys (like KeyClick and BootInfo) I need onyl to type one Tab :(
> 
> With the first Tab the (blue) ConfigDialog-Windows looses its focus (gets grey) and on the second Tab its gets the focus back and jumps to the next entry.

@fdivitto 
the commit [UI: TAB/SHIFT TAB inside open combobox selects right control](https://github.com/fdivitto/FabGL/commit/ad3fe588cb6cd04fde5421ae1e88e4d58f74dd06)
seem to work right with Shift-Tab (jumps from combobox to click box/combobox),
but forward with normal Tab it only jumps from Combobox to Combobox on my Confdialog-Menu.
So with Tab-forward I do never reach a Click-Entry :(
(But many thanks working so fast on that :) )