fdivitto/FabGL #183: DOS 6.22 isnt working anymore with 2 newest commits

guidol70, 2021-10-30:
I had MSDOS 6.22 german working with the master version of the 21.10.2021, but with the 2 latest commits (29.10/30.10) my german MSDOS 6.22 wont boot 
(only the Message "Starten von MSDOS" and hang) 
and your english MSDOS 6.22 will boot,but when executing EDIT or MSD 
it will get an error-message :(

fdivitto, 2021-10-30:
please can you post your german disk image?

guidol70, 2021-10-30:
> please can you post your german disk image?

you can get my german disk image [here](https://drive.google.com/file/d/17C0j7kr8CwssYYVuZFCCff5B1tGmWe4m/view?usp=sharing)

fdivitto, 2021-11-06:
thank you! Please could you using last commit?

guidol70, 2021-11-07:
> thank you! Please could you using last commit?

with the [newest i8086.cpp (from today)](https://github.com/fdivitto/FabGL/commit/95ae0a6a9d1a2cc3533273c43171f3523e2a0355) the german MS-DOS 6.22 start again ;)
Many Thanks

closed 2021-11-07 by guidol70