fdivitto/FabGL #184: Allow Terminal to interact with UI

anabolyc, 2021-11-14:
If I understood right, currently `Terminal` is able to capture keys from connected PS2 keyboard and send it away to `Serial` when it is attached to one via `Terminal.connectSerialPort(Serial)`. This is happening in asynchronously in `Terminal::keyboardReaderTask()` routine.

In my case I don't have PS2 keyboard, but instead i want to use connected Serial to send keystrokes to UI. So when `Terminal.pollSerialPort()` key is captured and propagated to `Terminal::write()` it should check if `m_keyboard` is present and push a key event via `m_keyboard->injectVirtualKey()`.

If implemented it would be possible to run UI library with full keyboard support without PS2 port, using Serial to PC connection only.

fdivitto, 2021-11-17:
In few words, do you need display on Terminal what is received from serial port? Is this like when you set "USBSerial" in AnsiTerminal application?

anabolyc, 2021-11-17:
Well, not really. Maybe I shouldn't mix in Terminal at all, let me explain.

What I'm looking for is to interact with UI components (and with PC emulator for the same reasons) using input methods other than PS2 devices, since they really hard to come by these days. 

Obvious candidate is serial console attached to ESP device, in other words my normal USB keyboard connected via laptop to ESP. In that case I would like to have some virtual PS2 device that would read Serial and translate it to UI components as keypress events. 

Since Terminal already has capability to bridge PS2 device with Serial, i thought it would be natural to use it to connect Serial to virtual PS2 device same way. But really it doesn't has to.

Another good example are physical buttons attached to ESP pins. Would be nice to bind physical buttons events to keypress events sent to UI as PS2 commands.

fdivitto, 2021-11-17:
I have to convert to discussion, because it is not strictly an issue. 
It is surely interesting, but a bit of work must be done. I imagine that on the PC side it should support ANSI escape code, to allow special keys to be recognised (cursor movement, function keys, ...), but I cannot imagine how SHIFT or ALT and so on can be passed over serial, using a simple terminal: you should think to a new protocol for this.
Also, once ANSI codes are received, they must be converted to scancodes, so Keyboard object can be used elsewhere (UI, Emulators, etc...). 
This is a bit of work...

closed 2021-11-17 by fdivitto