fdivitto/FabGL #195: TTGO T-display wont work at 40mhz spi bus

ken2228, 2021-12-24:
Building the simple rotate example for the ST7789 controller as on the Lily TTGO T-display gave me the following warning:

> E (732) spi_master: spi_bus_add_device(460): When work in full-duplex mode at frequency > 26.7MHz, device cannot read correct data.
>Try to use IOMUX pins to increase the frequency limit, or use the half duplex mode.
>Please note the SPI master can only work at divisors of 80MHz, and the driver always tries >to find the closest frequency to your configuration.
>Specify ``SPI_DEVICE_NO_DUMMY`` to ignore this checking. Then you can output data at >higher speed, or read data at your own risk

Reducing the TFT_SPI_WRITE_FREQUENCY in TFTControllerGeneric.cpp to 20000000 fixed this.

labrax, 2022-06-01:
Hi ken2228, I tried to get my Lily TTGO T-display to work, with no success. Could you please share your working code? Thank you

fdivitto, 2022-06-05:
commit https://github.com/fdivitto/FabGL/commit/c431ed3af707e647fe6292ac9094a22de5443086 adds support for TTGO T-Display (v1.1)

closed 2022-06-05 by fdivitto