fdivitto/FabGL #196: Offset on TTGO T-display

ken2228, 2021-12-25:
The ttgo display needs an offset in writing to RAM (RAMWR).
Adding/changing the following code in TFTControllerGeneric.cpp makes it work in Landscape mode (rotate 0%):

`void TFTController::sendScreenBuffer(Rect updateRect)`
`{`
 `int m_Coffset = 0;`
  `int m_Roffset = 0;`

  `if (m_screenWidth==135)`
  `{`
    `m_Coffset=52;`
    `m_Roffset=40;`
  `}`
  `....`
 `// Column Address Set`
  `writeCommand(TFT_CASET);`
  `writeWord(m_Coffset + m_rotOffsetX + updateRect.X1);   // XS (X Start)`
  `writeWord(m_Coffset + m_rotOffsetX + updateRect.X2);   // XE (X End)`

  `// Row Address Set`
  `writeCommand(TFT_RASET);`
  `writeWord(m_Roffset + m_rotOffsetY + updateRect.Y1);  // YS (Y Start)`
  `writeWord(m_Roffset + m_rotOffsetY + updateRect.Y2);  // YE (Y End)`

Using setRotation will result in the screen messing up.
Will need some more work to get it working in different rotations.

Keep up the good work!

Ken

k3v1nc, 2022-02-05:
I have the same issue on the M5StickC-Plus also using a 135x240 ST7789. The offset code above fixes the problem for me, thanks!

fdivitto, 2022-06-05:
commit https://github.com/fdivitto/FabGL/commit/c431ed3af707e647fe6292ac9094a22de5443086 adds support for TTGO T-Display (v1.1)

closed 2022-06-05 by fdivitto