fdivitto/FabGL #199: Keyboard not working with mouse, when no mouse the ESP32 reboot continuously...

exdiogene, 2022-01-16:
First I want to thank you for your fabulous library for VGA interface!

I have just installed the "Space Invaders" demo on my ESP32-WROVER chip with 4M SPRAM and the VGA part is working perfectly.

But when I connected my keyboard on the PS2 connector to Pins 32 and 33, it was not responding. I can only see the 3 lights flash when power is on.

When the keyboard is disconnected, I do not get the "Press [SPACE] to Play" message on screen, but it appear when connected.

I just tried to modify the code to disable the mouse, because I do not have one, with the following code : 

```
void setup()
{
//  PS2Controller.begin(PS2Preset::KeyboardPort0_MousePort1, KbdMode::GenerateVirtualKeys);
  PS2Controller.begin(PS2Preset::KeyboardPort0, KbdMode::GenerateVirtualKeys);

  DisplayController.begin();
  DisplayController.setResolution(VGA_320x200_75Hz);

  // adjust this to center screen in your monitor
  DisplayController.moveScreen(20, -2);
}

```
But it resulted with the ESP32 continuously rebooting after the code was transmitted!

When I code the ESP32, I must disconnect the VGA and keyboard to move it to my PC in another room.

I saw that you had fix the keyboard without mouse issue in 2020, could it have reappeared?

Thank you for your help,

 Exdiogene

exdiogene, 2022-01-22:
I finally found the reason of the issue!

When you have no mouse you cannot interrogate if the mouse is available or call any mouse function in the script...

In SpaceInvaders, because the PS2Controller is initialized with both keyboard and mouse, there is a call to see if the mouse is available and this disable the keyboard. If you just change the PS2Controller initialization to only the keyboard, you must take out any mouse verification in the script or it will reboot the CPU!

I believe that this behavior is not logical, if there is no mouse the script is suppose to continue with no mouse available without crashing and the keyboard must continue to work!

I hope that this issue would be fixed in a future release of the SpaceInvaders game.

exdiogene, 2022-01-24:
I finally found the reason of the issue!

When you have no mouse you cannot interrogate if the mouse is available or call any mouse function in the script...

In SpaceInvaders, because the PS2Controller is initialized with both keyboard and mouse, there is a call to see if the mouse is available and this disable the keyboard. If you just change the PS2Controller initialization to only the keyboard, you must take out any mouse verification in the script or it will reboot the CPU!

I believe that this behavior is not logical, if there is no mouse the script is suppose to continue with no mouse available without crashing and the keyboard must continue to work!

I hope that this issue would be fixed in a future release of the SpaceInvaders game.

There is the same issue in the VIC20 demo...

fdivitto, 2022-04-20:
I cannot replicate the first case (KeyboardPort0_MousePort1, with just the keyboard connected). Anyway now I fixed the code so it can work with just a keyboard or just a mouse.

Thanks!

closed 2022-04-20 by fdivitto