fdivitto/FabGL #205: Adding norwegian keyboard support to FabGL library

zakariassen, 2022-01-28:
Hello,

I have amended the files /src/devdrivers/kbdlayouts.h/.cpp to provide a Norwegian keyboard layout for FabGL.
As the norwegian alphabet has three additional letters compared to the US/UK alphabet, would I be correct to assume
that the Virtual key to string definitions for these chars also have to be added to keyboard.cpp? 

The letters in question are:

|  Glyph | html code         | Keyboard placement | scancode |
|--------|-----------------|----------------------|-----------|
| æ / Æ  | aelig; / AElig;     | OEM_7                     | 52           |
| ø / Ø   | oslash; /Oslash; | OEM_3                     | 4C           |
| å / Å    | aring; / Aring;   | OEM_6                     | 54            |

The letters are considered "full" letters, and not just accented versions of others.

As pushing files to the repo is not allowed, what is the preferred way to upload the amended files?

Regards,

Knut Zakariassen

zakariassen, 2022-02-01:
[kbdlayouts.cpp.txt](https://github.com/fdivitto/FabGL/files/7979729/kbdlayouts.cpp.txt)
[kbdlayouts.h.txt](https://github.com/fdivitto/FabGL/files/7979730/kbdlayouts.h.txt)

Uploading the amended files here, then we can take it from there.

fdivitto, 2022-04-09:
Thank you for this addition, and sorry for the delay. Could you suggest an ASCII name for the three virtual keys?
Something like VK_AELIG, VK_OSLASH, VK_ARING....but I am not sure if this is right!

zakariassen, 2022-04-10:
Ideally, since the chars are considered "full" letters in the norwegian alphabet and not accented version of others, the preferable ASCII names would be VK_æ, VK_ø and VK_å. 
In practice, though, going by the html name as you suggest probably will be easier to understand and use for non-norwegians. 

VK_aelig, VK_oslash and VK_aring are good "common-ground" names :)

GmEsoft, 2022-04-10:
Hi Knut,

As you should know, ASCII stands for American Standard Code for Information Interchange. It is a 7-bit coded character set, consisting in 32 control characters and 96 displayable characters, with the 26 Latin alphabet characters in upper and lower case, without accenting of any kind, the 10 digits and some punctuation and mathematical characters. The "preferable ASCII names" you suggest are not using plain ASCII chars, and are generally invalid to C++. The "common-ground" names are instead valid identifiers to C++.

Kind regards,

Michel Bernard

zakariassen, 2022-04-10:
Michel,

Indeed, and thanks for explaining better what I suspected all along. My profession is languages, not programming, so please forgive the misunderstanding.

Kind regards, Knut Zakariassen

fdivitto, 2022-04-10:
Ok, changes pushed. I haven't a Norwegian keyboard, so I cannot actually test it.

Thanks for the submission, and feel free to reopen the issue in case there are problems.

closed 2022-04-10 by fdivitto