fdivitto/FabGL #213: UART emulation?

majenkotech, 2022-02-22:
Do you have plans to create an emudev for the 16550 UART? This would massively enhance the PCEmulator and allow it to actually communicate with the outside world. It would be useful too if it were possible to choose where the serial connects to (physical UART, or a TCP socket, ideally with `ttynvt` compatibility).

fdivitto, 2022-06-23:
I'm working on 16450 and 8250 emulation (no FIFO). There are already some commits, so you can test it.

Something still doesn't work: for example I'd like to run LAPLINK to transfer files from PC to ESP32-PC, but I still can't do it!

At the moment the emulated UART is connected to USB serial connection.

majenkotech, 2022-06-29:
Awesome. Right now I am transferring msdos kermit zip file using xmodem in the windows 3 terminal at 9600 baud :)

fishhack66, 2022-07-05:
> Awesome. Right now I am transferring msdos kermit zip file using xmodem in the windows 3 terminal at 9600 baud :)

Hi guys.  Majen - could you send me a PM; I have some questions about establishing connection via the Windows 3.0 Terminal (mainly to do with hardware cabling...need a null modem, some USB size adapters??), and I don't want to bog down this issue thread.  Thanks, Dave

majenkotech, 2022-07-07:
Incidentally, the best way I have found of transferring files so far is using XModem with Procomm on the DOS side and Minicom on the Linux side. I'm just in the process of seeing if it's possible to use epppd and mTCP to connect DOS to the network :)

fdivitto, 2022-07-13:
There is a free and nice tool I'm using on both sides (real PC and ESP32 emulated), to transfer files, using a UI like laplink. It is File Maven:

http://www.briggsoft.com/fmdos.htm

If you run it on an emulated PC (like dosbox) you should slow down it to 8Mhz otherwise it could not reliability  communicate.

Of course there other ways: I like for example c-kermit, which runs on almost all OS and platforms (included CP/M).

closed 2022-08-27 by fdivitto