fdivitto/FabGL #227: [Question] different Screen-Height on 640x480x60 with VGA8-/VGA16-Controller

guidol70, 2022-04-11:
Thanks to some of your latest changes now I do get a picture/sync **with 
ESP32-Core v2.x.x** and the VGA16-Controller 

To get get 640x480x60 (for 80x25 chars) I do use 
```
fabgl::VGA8Controller   DisplayController;
or
fabgl::VGA8Controller   DisplayController;
and
DisplayController.begin(); 
DisplayController.setResolution(VGA_640x480_60Hz);
```
but with the VGA16-Controller the Screen-Height it seems that the 25 Lines are NOW more compressed than with the VGA8-Controller.

For both Controllers my Monitor dows show as Info 31.5KHz, 60Hz 
and 640x480

Do I miss something in my code or does have the VGA16-Controller now some other timing. As I recall from the past there were no difference in the Screen-Height.

I think the setfont-Command (for 8x19) would only be availabe in the VGATextController?
[image]
[image]
[image]
[image]

fdivitto, 2022-04-11:
Please could you tell me (using a printf or terminal output), the value returned by method DisplayController.getViewPortHeight(), in both cases (8 and 16 colors)?

guidol70, 2022-04-11:
> printf  value returned by method DisplayController.getViewPortHeight(), in both cases (8 and 16 colors)

printf says for
getViewPortHeight for  8 Colors = 480
getViewPortHeight for 16 Colors = 388

fdivitto, 2022-04-11:
yes, it can just allocate 388 of 480, so it cannot allocate 92 lines. Note that 92 lines require 92*640/2= 30 KBytes.

Current version has a new paramter in fabglconf.h, named FABGLIB_MINFREELARGESTBLOCK. It contains the largest block of DMA memory that must be kept free.
You can try to lower this value (for example to 10000)...of course other stuff may not work if you prioritise framebuffer allocation.

guidol70, 2022-04-11:
> FABGLIB_MINFREELARGESTBLOCK. 
You can try to lower this value (for example to 10000)

with 10000 the VGA32 only resets in a loop
with 20000 I do get 452 lines

> A question: could you tell me the free memory just before "DisplayController.setResolution" ?

before setresolution the memory looks like
Free DMA Memory            ->   193396 bytes
Free Memory                ->   252356 bytes

with ESP32-Core v1.0.6
before setresolution the memory looks like
Free DMA Memory            ->   215024 bytes
Free Memory                ->   288488 bytes

I do get 480 Lines and my rest-memory is
DEBUG : Free DMA Memory     [ 14748 Byte   ]
DEBUG : Free Memory         [ 86276 Byte   ]

So - in the end I will go with 8 colors to get no memory problems with ESP32-Core v2.x.x :(

But another lesson learned...

fdivitto, 2022-04-12:
well, so we have lost about 36K with 2.x versions.. too bad! I don't know how to recover this memory just with optimizations!

I guess 20000 is a good one then.

guidol70, 2022-04-12:
> I guess 20000 is a good one then.

Hmm - but then free DMA Memory was only around 7.700 bytes :(
I doesnt feel "safe" with that amount ;)

fdivitto, 2022-04-12:
> Hmm - but then free DMA Memory was only around 7.700 bytes :( I doesnt feel "safe" with that amount ;)

Setting 20000 means you will have 20K free anyway. DMA could be useful (I guess) only in special cases (WiFi? BLE?).
However I am thinking to set this parameter at runtime, because it is actually an application specific parameter.

fdivitto, 2022-04-12:
also, why not use an ESP32 with PSRAM for your emulator? You'll forget all RAM problems (think to my PC IBM Emulator that runs without problems at 640x480...)

guidol70, 2022-04-14:
> also, why not use an ESP32 with PSRAM for your emulator? 

for the VGA32-Port - ther is already 8MB PSRAM on the VGA32, but I dont know how to mak euse of it ;)

For the Memory-setting of 20000:
Do you know / is there any way to get rid of the Wifi/BLE-Code, because on the VGA32 and this version I dont use it (I only had once a telnet-server-version where I used WiFi)