fdivitto/FabGL #229: [Question] Input/Output-problems with FabGL Terminal and RC2014-Emulation

guidol70, 2022-04-18:
Hi,
maybe you got an idea for me?
I ported the a CutDown/Junior-Version of the [ESP32-Z80-Emulator](https://github.com/djbottrill/ESP32-Z80-Emulator) to a TTGO VGA32 which does use the  FabGL-Library and a locally connected Terminal for VGA-Output.

Now I want to track down some issues:

Input-Problem:
On some circumstances I have to press Enter/Return twice to get it recognized.
The first time after start is after the Emulator displays "Starting Z80".
There I have to press enter to get a prompt.
At the A> prompt when I only press Enter or press Enter after some Spaces it does work.
When I give the DIR command the Cursor does go to the start of the Line after the first time I press Enter (LF?) and after the second time I press Enter (CR?) it starts the command.

Output-Problem:
I could do give the command DIR serveral times (Ok with pressing twice Enter) and the output works fine.
But when I start a BASIC-program with a Mandelbrot/Fractal which should output many lines it stops after some lines and I have to press any key to get further display-output.

Now I think its a problem (maybe Iam using it wrong?) with the local Terminal, because when I reroute (via the source) the output to the USB-serial port then everything works fine - Input and Output.

Is there some kind of Handshake useable with the VGA32? I didnt got this in my RunCPM-Port (where I also use the same ConfigDialog) - I copied the Terminal part from my RunCPM-Source.

My Source (if you mind to take a look) is at
https://github.com/guidol70/RC2014_ESP32_Z80_Emulator_VGA32
[image]

fdivitto, 2022-04-18:
please could you replicate this problem with one of the FabGL examples?

fdivitto, 2022-04-19:
I have to close until it is not replicable with FabGL examples.

closed 2022-04-19 by fdivitto