fdivitto/FabGL #242: Unable to include "fabgl.h" in platformio project

yuanb, 2022-05-23:
Not sure if I am the only one has this project. 

My environment: Mac OS X 10.15.7, VS Code 1.67.2, platformIO extension code 6.0.1, Home 3.4.1 and Espressif 32 version 4.3.0

I created a hello-world project, after adding 'FabGL' library 1.0.6 to my project. The platform.ini is like this:
```
[env]
monitor_speed = 115200
monitor_flags = --raw

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
lib_deps = 
	fdivitto/FabGL@^1.0.6
```

I'm unable to use to #include "fabgl.h". I'm getting this error
```
Building in release mode
Compiling .pio/build/esp32dev/src/main.o
src/main.cpp:8:10: fatal error: fabgl.h: No such file or directory

***************************************************************
* Looking for fabgl.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:fabgl.h"
* Web  > https://registry.platformio.org/search?q=header:fabgl.h
*
***************************************************************

 #include "fabgl.h"
                 ^~~~~~~
```

I then opened the fabgl.h in project/.pio/libdeps/esp32dev/FabGL/ folder. There are many headers in this files shown /w 'red' dashes. 

Am I missing anything? I apologize if this has been asked before.

Thanks

fdivitto, 2022-05-23:
please try version 1.0.7 instead of 1.0.6

yuanb, 2022-05-23:
Upgraded to 1.0.7, same error:

~~~
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.19.4
Building in release mode
Compiling .pio/build/esp32dev/src/main.o
src/main.cpp:8:10: fatal error: fabgl.h: No such file or directory

***************************************************************
* Looking for fabgl.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:fabgl.h"
* Web  > https://registry.platformio.org/search?q=header:fabgl.h
*
***************************************************************

 #include "fabgl.h"
          ^~~~~~~~~
compilation terminated.
*** [.pio/build/esp32dev/src/main.o] Error 1
================================================================================================== [FAILED] Took 28.89 seconds ==================================================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
~~~

fdivitto, 2022-05-23:
to include a c++ header (as is fabgl.h) you may rename to cpp the main, but in this case you have to export "app_main" with C linkage, for example:

```
#include "fabgl.h"

extern "C" {
    void app_main(void);
}

void app_main() {
  InputBox ib;
  ib.begin();
  ib.message("", "Hello World!");
}
```

yuanb, 2022-05-23:
Thanks for your reply, with 1.0.7 release:
I created a second project to use Arduino framework and added  "lib_ldf_mode = deep+" to platformio.ini.
it compiled.

My first project was created with "Espressif IoT Development framework". Yes, the app_main is declared with extern "C". I understand you said the IDF framework is not supported yet.

fdivitto, 2022-05-23:
I've just issued version 1.0.8 for this purpose, locally it works with IDF, but I am waiting PlatformIO update to be sure.

fdivitto, 2022-05-23:
...this is the message I've removed waiting for 1.0.8....
to include a c++ header (as is fabgl.h) you may rename to cpp the main, but in this case you have to export "app_main" with C linkage, for example:

```
#include "fabgl.h"

extern "C" {
    void app_main(void);
}

void app_main() {
  InputBox ib;
  ib.begin();
  ib.message("", "Hello World!");
}
```

yuanb, 2022-05-23:
> I've just issued version 1.0.8 for this purpose, locally it works with IDF, but I am waiting PlatformIO update to be sure.

Thanks. With my project switched to Arduino framework. I compiled DoubleBuffer, it worked. I will continue my work and switch to IDF framework when your code is ready.

yuanb, 2022-05-23:
In 1.0.8, I think you missed "," at the end of line #5 of library.json
[image]

fdivitto, 2022-05-23:
you're right, thanks!

yuanb, 2022-05-23:
I can confirm, with the "," issue fixed in 1.0.8 FabGL library. I am able to build and run "SimpleTerminalOut" example with IDF framework project with PlatformIO in VSCode.

fdivitto, 2022-05-24:
> I can help to port your Arduino examples to IDF if you think there is a need. Most will be straightforward. Yes?

yes, but not now. Before this I'd like to embed into FabGL several functionalities now coded inside examples, so examples will be reduced to just few lines of code.

closed 2022-05-24 by fdivitto