fdivitto/FabGL #248: Multitasking CPM Plus - visible VGA screen corruption and screen crashes

homebrew-z80, 2022-06-20:
With one console doing NULU and extraction of large LBR file, and second  console doing minor tasks I have experienced some crashes and visible corruption on the active screen (e.g. memory leak?)

Also on one occasion I had an out of memory error, with just 2 or 3 consoles, where system should have enough memory for that.

sdcard, v1.08 fabgl,  fabgl dev board v4 , esp32 huge app

fdivitto, 2022-06-23:
how much free memory you have on both consoles (run "info" command)?

homebrew-z80, 2022-06-30:
I apologise for being very slow getting back to you.

I have tried for several days now to reproduce the VGA screen corruption
error and have not been able to achieve it. I have used my ESP32 dev board
v3 and my VGA32 v1.4

The visual corruption was  of the nature of wrong video attributes for the
characters on the top few lines of screen, showing as character blobs. I
mention this because it is not a symptom of a poor vga connection.

In my first attempt when I reported the error I was only using up to 5 or 6
terminals, but have subsequently been successful with all 12, not actively
running programs in all, but was running programs in 3 and switching
through all 12.

I shall continue to use it with a broader range of CP/M programs and see if
anything re-appears, and I am keen to move to primarily using a serial
connection.

I wlll update the github bug report  accordingly.

Thank you for such an interesting set of examples, excellent for a CP/M
hobbyist.

Also, as I am in the UK, not supported by Tindie due to stupid Brexit, is
there a method by which I can purchase your newer "FabGL ESP32 Board 16MB
Flash 4 MB PSRAM - 3.3V I/O"?

Regards
Richard

fdivitto, 2022-06-30:
Memory for each session is dynamically allocated and sometimes it could be not enough to run all sessions. Please try to flash the Multitasking Sketch enabling PSRAM on VGA32 1.4: this should avoid some memory problems.

homebrew-z80, 2022-10-11:
Thanks, I will try that.
Richard