fdivitto/FabGL #258: Mount Disk fails in PCEmulator

majenkotech, 2022-07-07:
Using the Alt-SysRQ / Mount Disk option doesn't appear to mount the disk (or doesn't inform the system the disk has changed). As soon as you mount a disk all subsequent accesses respond with
```
Not ready reading drive A
Abort, Retry, Fail?
```

This makes it impossible to install anything that is on multiple disk images (including DOS 6.22)

fdivitto, 2022-08-27:
I'm having some difficulties to fix this problem, so I ask you to test my last commit (just replace the full library and examples with current repo content).
If you are continuing to have this error, please let me know from which image do you start (freedos from floppy, or msdos...)

thanks for the tests

majenkotech, 2022-10-01:
It doesn't matter about what image you use. If you start from a blank disk and try installing MS-DOS 6.22 you see the problem (you can't install as you can't change disks).  I'm just setting it up again now (I have since got a real 486 for DOS stuff so I don't really need the emulator now, but it's still fun to play with) to try it out.

From my understanding a drive should de-assert its ready signal when the door opens or a disk is ejected and it should stay de-asserted until the next STEP signal, at which point if a disk is there it should assert its ready signal.

majenkotech, 2022-10-01:
Whatever it is you've done, though, seems to have cured the problem. I am now on disk 2 of DOS 6.22 and installing happily.

closed 2022-10-01 by majenkotech