fdivitto/FabGL #277: Modeline for 1024x600 display

drtjs, 2022-09-19:
Hi Fabrizio,

I'd love to attach one of my 7 inch 1024x600 TFT displays to my VGA32_1.4 and utilize the native resolution, but it seems I can't :-(

Unfortunately, there is no corresponding predefined modeline entry in fabglconf.h.
So I tried several modeline parameter sets I could find in various corners of the internet, but I wasn't lucky: The Display didn't sync to any of these parameter sets while it synced easily to other prdefined resolutions.
I then attached the display to an old Dell Latitude 6320 running a slightly aged new copy of System Rescue (SystemRescueCD). SysRecCD immediately detected the display and configured it properly using the display's native resolution. Based on the output of xrandr --verbose on the Latitude notebook I generated a new modeline:
Modeline "1024x600@60Hz" 49 1024 1041 1185 1312 600 601 604 624 +HSync +VSync
I added this to fabglconf.h and a corresponding entry to ModelineStudio. This time I was partially successful:
ModelineStudio output for this entry:

Modeline "1024x600@60Hz" 49 1024 1041 1185 1312 600 601 604 624 +HSync +VSync  VisibleBegins
moveScreen(-1, 0)
Screen size   : 1024 x 600
Viewport size : 1024 x 600
Free memory (total, min, largest): 208576, 174416, 110580

[image]

It takes several tens of seconds to achieve a very flickery sync as can be seen in the picture included above.

I' m afraid I'm lost. Any ideas w.r.t. what is going wrong, here? Are there any fundamental problems I' not aware of? In case there is hope ;-) I would be happy to provide additional information or perform tests...

All the best and carry with this great project,

Thomas

tsacks, 2023-03-03:
I have a dang similar screen with the same resolution, and am running into the same issue. Busted out an old laptop with VGA that I could boot linux on, to see if I could see what Modeline it's using to run the output, and ended up with these settings, very close to yours:

"1024x600@59.85" 49 1024 1040 1184 1312 600 601 604 624 +HSync +VSync

It's still doing the wobbly thing

tsacks, 2023-03-03:
Just to add some info to this one, I pulled the EDID data from the monitor over the VGA port, and those are definitely the timings that it's reporting should be using. At least, if my math checks out.

Decoded EDID data:
```
Detailed Timing Descriptor
Pixel Clock: 49MHz
Horizontal Active: 1024
Horizontal Blanking: 288
Vertical Active: 600
Vertical Blanking: 24
Horizontal Sync Offset: 16
Horizontal Sync Pulse: 144
Vertical Sync Offset: 1
Vertical Sync Pulse: 3
Horizontal Display Size: 337
Vertical Display Size: 270
Horizontal Border: 0
Vertical Border: 0
Interlaced: false
Stereo Mode: 0
Sync Type: 3

2-Way Line-Interleaved Stereo: false
```

Raw EDID data:
```
00ffffffffffff006a991019502d3101
0a17010308211b78eaeea69e544c9926
194f54afcf0061400101010101010101
01010101010124130020415818201090
1300510e1100001e000000fc004c4d31
3731302028564741290a000000fd0037
4c1e510e000a202020202020000000ff
0032303030303038300a2020202001eb
00ffffffffffff006a991019502d3101
0a17010308211b78eaeea69e544c9926
194f54afcf0061400101010101010101
01010101010124130020415818201090
1300510e1100001e000000fc004c4d31
3731302028564741290a000000fd0037
4c1e510e000a202020202020000000ff
0032303030303038300a2020202001eb
```