fdivitto/FabGL #279: is it possible output 1024X768 with 64 colors?

leodefpv, 2022-09-24:
hi 
your lib is so great and easy to use
thanks for your great job that i can use esp32 out put VGA singal 
i want to use esp32 out put 1024X768 VGA singal with 64 color
but it can't show all Viewport. At first i though it's memory problem
so i change esp32 to WROVER type with PSRAM
but it still not show all view 
and i print info like this

------------------------------------------------------------------------------
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
Screen size   : 1024 x 768
Viewport size : 1024 x 179
Free memory (total, min, largest): 4268947, 4261043, 4128756
------------------------------------------------------------------------------

it doesn't appear to be using all available memory。
how can i use PSRAM to add Viewport size?
Or it can't output 1024X768 with 64 color?

Thanks for your attention and reply！

fdivitto, 2022-09-24:
PSRAM is not usable for DMA output

leodefpv, 2022-09-24:
so it's impossible output 1024 X 768 with 64 color ?

fdivitto, 2022-09-24:
64 colors require 6 bits, so 1024x768x6 = 4718592 bits = 589824 bytes = 576 KB, that is out of ESP32 specs!

leodefpv, 2022-09-24:
Thank you for your answer
if change the chip to esp32 S2 or S3 
can the EDMA do this job to output 1024X768?

fdivitto, 2022-09-24:
> can the EDMA do this job to output 1024X768?

I'm sorry, I don't know! 
probably not, because PSRAM could not provide enough speed for 1024x768 dot frequency...or who knows what else kind of problem!

leodefpv, 2022-09-24:
thank you!
maybe S3 can do this i will try .

leodefpv, 2022-09-25:
Esp32 S3 PSRAM speed is 12Mhz 
Is this fast enough to drive VGA?

fdivitto, 2022-09-25:
1024x768 requires 44Mhz at least. Look at this for reference:

[http://tinyvga.com/vga-timing](http://tinyvga.com/vga-timing)

defdefred, 2022-11-09:
I wonder if the "wifi" core of ESP32 could be hacked to become a more powerfull fabgl vga graphic card?