fdivitto/FabGL #288: PC Emulator: map SD folder as Floppy 1?

vitasam, 2022-10-27:
Hello,
what about automatic mapping of selected  folder from SD-card as a Floppy drive? (e.g  in a similar way DosBox does)

badre2911, 2022-11-28:
It's a good idea, I hope it will be doable.

niutech, 2023-05-03:
You can already map a disk image from SD card as a floppy A: or B: (using `Machine::setDriveImage()` internally).

vitasam, 2023-05-07:
I see. Although I meant the usual folder on the SD card (visible from a PC if you insert the card into the reader), rather than the image: `Machine::setDriveImage(int drive, char const * filename, int cylinders, int heads, int sectors)`

niutech, 2023-05-07:
As a workaround, you can mount the disk image file as a drive on a PC using OSFMount, ImDisk or Disk Image Mounter.