fdivitto/FabGL #301: FabGL dev.board missing I/O Expander

vitasam, 2022-11-16:
Hello,
I was trying [GPIOGW.BAS](https://github.com/fdivitto/FabGL/blob/master/Extras/PC%20Emulator%20BASIC%20Programs/GPIOGW.BAS) on the latest FabGL board (3.3v) I bought recently, but it returns following:
`Extended I/O not available on this board!`

Is this BASIC program is not relevant for the board? MCP23S17 is installed.

fdivitto, 2022-11-17:
Hello,
there is an error in GWBASIC version of that program (I didn't remember that IF..THEN in GWBASIC must be in the same line!!). Please test the [new](https://github.com/fdivitto/FabGL/blob/master/Extras/PC%20Emulator%20BASIC%20Programs/GPIOGW.BAS) version.

Thanks!

vitasam, 2022-11-17:
Thanks! Interesting, I did not remember about this in GWBASIC too. 
The new code works well.

closed 2022-11-17 by vitasam