fdivitto/FabGL #330: ESP32-S3 support -  soc/frc_timer_reg.h missing?

UKTailwind, 2023-04-15:
I've tried to compile the simplest code for a range of the S3 boards and in each case I get the same error with the compilation unable to find soc/frc_timer_reg.h
There is no issue if I choose a ESP32-DEVKIT-V1 or many other of the earlier boards

`-DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\peter\\AppData\\Local\\Temp\\arduino_build_721824/build_opt.h" "-IC:\\Users\\peter\\AppData\\Local\\arduino15\\packages\\esp32\\hardware\\esp32\\2.0.7\\cores\\esp32" "-IC:\\Users\\peter\\AppData\\Local\\arduino15\\packages\\esp32\\hardware\\esp32\\2.0.7\\variants\\esp32s3" "-ID:\\peter\\Documents\\Arduino\\libraries\\FabGL\\src" "C:\\Users\\peter\\AppData\\Local\\Temp\\arduino_build_721824\\sketch\\EspMite.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for soc/frc_timer_reg.h: []

ResolveLibrary(soc/frc_timer_reg.h)

In file included from D:\peter\Documents\Arduino\libraries\FabGL\src/fabgl.h:312,  -> candidates: []

                 from D:\peter\Documents\Arduino\EspMite\EspMite.ino:1:

D:\peter\Documents\Arduino\libraries\FabGL\src/fabutils.h:45:10: fatal error: soc/frc_timer_reg.h: No such file or directory

 #include "soc/frc_timer_reg.h"

          ^~~~~~~~~~~~~~~~~~~~~

compilation terminated.

Using library FabGL at version 1.0.9 in folder: D:\peter\Documents\Arduino\libraries\FabGL 

exit status 1

Error compiling for board ESP32S3 Dev Module.

`

ericjenott, 2023-09-20:
I am having the same issue.  Did you ever find a solution to this?

endrew74, 2023-09-20:
https://github.innominds.com/fdivitto/FabGL/discussions/312

endrew74, 2023-09-22:
https://github.com/OliviliK/ESP32_timer_u32

soporteimatt22, 2023-10-24:
were you able to solve it? The same thing appears to me with an 8MB esp32-s3 n8r8. Trying to use any of the examples, errors appear everywhere.

soporteimatt22, 2023-10-24:
> I am having the same issue. Did you ever find a solution to this?

work?

endrew74, 2023-11-10:
I haven't tested it, but it should work. We need to understand how to replace one with the other. The point is that esp32 and esp32s3 have different system timers. Accordingly, the interfaces are different.

jonathan0170, 2024-03-19:
Have been working with FABGL VGA with Arduino IDE.
After a few months break, Arduino upgrades, now get compilation error, similar to above:

...Arduino\libraries\FabGL\src/fabutils.h:45:10: fatal error: soc/frc_timer_reg.h: No such file or directory

Cannot compile ant FABGL code? using

jonathan0170, 2024-03-19:
Have reinstalled FABGL. no change.
Related to fabutils.h
[image]

Can anybody assist?

jonathan0170, 2024-03-20:
Thanks for response..
Have been trying for a while to get any advance on this one.
Is there a better way to compile FABGL library code other than Arduino IDE that may circumvent the issue?