fdivitto/FabGL #349: Uzebox-alike open source console platform

danboid, 2023-07-28:
Hi Fab

I had great fun building and writing a game for the open source Uzebox games console. Today I got to thinking about what sort of spec and form a Uzebox successor, or at least another fully open source console platform like the Uzebox might have and thats how I've ended up here when I found FabGL.

I see there are a few boards which are capable of running FabGL but most of them don't see to be available any more except for the Lilygo FabGL VGA32. Is there a better board for running FabGL thats currently available to order from the UK?

None of the boards I see listed on https://github.com/fdivitto/FabGL/wiki/Boards seem to be intended for gaming, non of them feature any joystick ports unless you can get a Bluetooth game controller working. 

Have you considered creating an open source console and API similar to the Uzebox by creating a board with a couple of SNES or Playstation style controller sockets and creating a simple bootloader/menu for loading games off an SD card? It could be called the FabBox or the FaBoX, maybe?

mobluse, 2023-07-29:
There is ESP32-SBC-FabGL in UK https://thepihut.com/products/olimex-esp32-sbc-fabgl and also from the original place: https://www.olimex.com/Products/Retro-Computers/ESP32-SBC-FabGL/open-source-hardware . I don't own one and still only have TTGO VGA32 V1.4 (€15/$15 including shipping).

ESP32-SBC-FabGL can use one Nunchuck (but needs an adapter): https://olimex.wordpress.com/2023/06/27/esp32-sbc-fabgl-update-now-you-can-play-invaders-with-wii-nunchuck-via-the-uext-port/

There is a port splitter that might work with 4 Nunchucks, but I cannot guarantee that:
https://www.olimex.com/Products/Modules/Interface/UEXTx5/

Of course that would need software support that doesn't exist today.

WII NUNCHUCK CONTROLLER WITH UEXT CONNECTOR:
https://www.olimex.com/Products/Modules/Sensors/MOD-WII/MOD-Wii-UEXT-NUNCHUCK/open-source-hardware

Otherwise many PC gamers are happy with just keyboard and mouse, and that works now.

danboid, 2025-10-16:
Hi @mobluse 

Thanks for your reply, which I only spotted today somehow.

Has anyone created a UEXT to (6 button) Mega Drive controller adapter, that you've heard about?

It seems FabGL is dead? Is there an active fork?

What is the best ESP32 model to use with FabGL or a fork now?

randybb, 2025-10-16:
Would be nice to switch to esp32-s3 or even esp32-p2 (+c6 for wifi). Just there are no ruch boards with vga