fdivitto/FabGL #350: enhancement: add the Termcap Library

snakebyte69, 2023-07-29:
Add the Termcap Library so any terminal can be used
https://www.gnu.org/software/termutils/manual/termcap-1.3/html_mono/termcap.html

mobluse, 2023-07-29:
I believe Termcap is a server-side library, and FabGL is mostly a client library.

geneb, 2023-08-11:
termcap is used to tell a host computer what escape sequences to send to a terminal for the various cursor control functions.  This allows a program written for terminal use to not have to rely on hard-coded escape sequences for a specific terminal emulation - it can simply make a curses (or ncurses) call for a cursor movement operation and the api will figure out what escape sequence needs to be set based upon the termcap entry for the terminal being used.