fdivitto/FabGL #388: Failed to compile examples of FabGL

FernandoRoxo, 2024-07-11:
Terrible title, I know.  Sorry.

I have Arduino IDE 2.3.2 installed, installed FabGL 1.0. using the Library Manager and installed esp32 3.0.2 by espressif as instructed by many videos on internet  (preferences->Additional boards manager...).

The Arduino IDE is the 'arduino-ide_2.3.2_Linux_64bit.AppImage' on Linux Ubuntu 20.04.

I selected the ESP32 Dev Module and started compiling the example sketches from FabGL.

The first I have tried was the VIC20 with the partition Scheme Huge App. The compilation finished with the error:

```
/tmp/.arduinoIDE-unsaved2024611-1038633-wjo2zo.5ue8c/VIC20/VIC20.ino:38:17: note: '#pragma message: This sketch requires Tools->Partition Scheme = Huge APP'
   38 | #pragma message "This sketch requires Tools->Partition Scheme = Huge APP"
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/.arduinoIDE-unsaved2024611-1038633-wjo2zo.5ue8c/VIC20/src/machine.cpp: In member function 'void Machine::syncTime()':
/tmp/.arduinoIDE-unsaved2024611-1038633-wjo2zo.5ue8c/VIC20/src/machine.cpp:246:5: error: 'ets_delay_us' was not declared in this scope; did you mean 'sys_delay_ms'?
  246 |     ets_delay_us(delayNS / 1000);
      |     ^~~~~~~~~~~~
      |     sys_delay_ms

exit status 1
```

Compilation error: 'ets_delay_us' was not declared in this scope; did you mean 'sys_delay_ms'?

I tried to compile the Atari8800 and it finished with:
```
/home/roxo/Arduino/libraries/FabGL/src/comdrivers/tsi2c.cpp: In static member function 'static void fabgl::I2C::commTaskFunc(void*)':
/home/roxo/Arduino/libraries/FabGL/src/comdrivers/tsi2c.cpp:232:89: error: invalid conversion from 'uint32_t*' {aka 'long unsigned int*'} to 'size_t*' {aka 'unsigned int*'} [-fpermissive]
  232 |       job->lastError = i2cRead(i2c, job->address, job->buffer, job->size, job->timeout, &job->readCount);
      |                                                                                         ^~~~~~~~~~~~~~~
      |                                                                                         |
      |                                                                                         uint32_t* {aka long unsigned int*}
In file included from /home/roxo/.arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/esp32-hal.h:80,
                 from /home/roxo/Arduino/libraries/FabGL/src/comdrivers/tsi2c.h:40,
                 from /home/roxo/Arduino/libraries/FabGL/src/comdrivers/tsi2c.cpp:33:
/home/roxo/.arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/esp32-hal-i2c.h:36:114: note:   initializing argument 6 of 'esp_err_t i2cRead(uint8_t, uint16_t, uint8_t*, size_t, uint32_t, size_t*)'
   36 | esp_err_t i2cRead(uint8_t i2c_num, uint16_t address, uint8_t *buff, size_t size, uint32_t timeOutMillis, size_t *readCount);
      |                                                                                                          ~~~~~~~~^~~~~~~~~
/home/roxo/Arduino/libraries/FabGL/src/comdrivers/serialport.cpp: In member function 'void fabgl::SerialPort::setup(int, uint32_t, int, char, float, fabgl::FlowControl, bool)':
/home/roxo/Arduino/libraries/FabGL/src/comdrivers/serialport.cpp:241:3: error: 'gpio_matrix_in' was not declared in this scope
  241 |   gpio_matrix_in(m_rxPin, URXD_IN_IDX[m_idx], m_inverted);
      |   ^~~~~~~~~~~~~~
/home/roxo/Arduino/libraries/FabGL/src/comdrivers/serialport.cpp:242:3: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
  242 |   gpio_matrix_out(m_txPin, UTXD_OUT_IDX[m_idx], m_inverted, false);
      |   ^~~~~~~~~~~~~~~
      |   gpio_iomux_out
/home/roxo/Arduino/libraries/FabGL/src/comdrivers/serialport.cpp: In member function 'void fabgl::SerialPort::sendBreak(bool)':
/home/roxo/Arduino/libraries/FabGL/src/comdrivers/serialport.cpp:343:5: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
  343 |     gpio_matrix_out(m_txPin, MATRIX_DETACH_OUT_SIG, m_inverted, false);
      |     ^~~~~~~~~~~~~~~
      |     gpio_iomux_out
/home/roxo/Arduino/libraries/FabGL/src/comdrivers/serialport.cpp:347:5: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
  347 |     gpio_matrix_out(m_txPin, UTXD_OUT_IDX[m_idx], m_inverted, false);
      |     ^~~~~~~~~~~~~~~
      |     gpio_iomux_out

exit status 1

Compilation error: exit status 1
```

I have tried to compile  the following: 

ANSI Terminal, Audio, ClassicRacer, ColisionDetection, DirectVGA, Hardware Test and Loopback Terminal

All of them finished the compilation with errors, the first one being:
```
/home/roxo/Arduino/libraries/FabGL/src/comdrivers/tsi2c.cpp: In static member function 'static void fabgl::I2C::commTaskFunc(void*)':

/home/roxo/Arduino/libraries/FabGL/src/comdrivers/tsi2c.cpp:232:89: error: invalid conversion from 'uint32_t*' {aka 'long unsigned int*'} to 'size_t*' {aka 'unsigned int*'} [-fpermissive]
  232 |       job->lastError = i2cRead(i2c, job->address, job->buffer, job->size, job->timeout, &job->readCount);
      |                                                                                         ^~~~~~~~~~~~~~~
      |                                                                                         |
      |                                                                                         uint32_t* {aka long unsigned int*}
   ```   
     
      Am I missing something?   Perhaps the version of the Arduino IDE?
      
      If this in not the forum for this, please let me now and, if possible, point  me to the correct one.
      
      Thanks in advance.

FernandoRoxo, 2024-07-13:
I have moved this to the discussions, seems to fir better there.
I couldn't find a way to  close or remove this issue.   

😒

Xalior, 2024-07-18:
I get this when using v3 of the esp32 board library in the arduino SDK, if you pick the final version of v2 (2.0.17 for me) the examples no longer produce this error.

Good Luck!

FernandoRoxo, 2024-07-19:
Hi Xalior,

Thank you very much.   I have installed the version 2.0.17 and the examples compiled without problem

closed 2024-07-19 by FernandoRoxo