fdivitto/FabGL #393: Compilation Error: Missing WiFiGeneric.h in ICMP.cpp when using FabGL with ESP32

YonLiud, 2024-08-17:
I'm encountering a compilation error when running a basic program using FabGL on an ESP32 with the PlatformIO environment (using the Arduino framework). The error occurs when the compiler is unable to find the WiFiGeneric.h header file.

I attempted to add third-party libraries that include the WiFiGeneric.h header, but this did not resolve the issue.

```cpp
#include "fabgl.h"

fabgl::VGAController DisplayController;
fabgl::Terminal      Terminal;

void setup() {
    DisplayController.begin();
    DisplayController.setResolution(VGA_400x300_60Hz);
    Terminal.begin(&DisplayController);
    Terminal.println("Hello, World!");
}

void loop() {
}
```

The logs returned:

```
Compiling .pio\build\esp32doit-devkit-v1\lib391\FabGL\network\ICMP.cpp.o
.pio/libdeps/esp32doit-devkit-v1/FabGL/src/network/ICMP.cpp:33:10: fatal error: WiFiGeneric.h: No such file or directory

*********************************************************************
* Looking for WiFiGeneric.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiGeneric.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiGeneric.h
*
*********************************************************************

 #include "WiFiGeneric.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib391\FabGL\network\ICMP.cpp.o] Error 1
```

YonLiud, 2024-08-17:
I found a temporary solution to the issue. By modifying the include directive in ICMP.cpp directly:
```diff
#include "ICMP.h"

#include "Arduino.h"
- #include "WiFiGeneric.h"
+ #include "C:/Users/<user>/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h"

namespace fabgl {
```

the code compiles successfully. However, this is not an ideal solution as it involves hardcoding the full path to the file. A more permanent fix would be preferable.

BuslAV, 2024-10-11:
Hello,

I faced an issue while working on the FabGL project where the error message fatal error: WiFiGeneric.h: No such file or directory appeared. To resolve this issue, I found that it was necessary to explicitly include the WiFi library in the project dependencies.

Here’s how I fixed it:

In the platformio.ini file, I added WiFi to the lib_deps section to ensure that the WiFi library is included:

lib_deps = 
    fdivitto/FabGL@^1.0.9
    WiFi

Additionally, make sure to include the WiFi header in your main source file:
#include <WiFi.h>

By making these adjustments, the error related to WiFiGeneric.h was resolved, allowing the project to compile successfully.

Thank you for your attention to this issue!