fdivitto/FabGL #407: Does the FABGL library support UTF-8 on FABGL board ?

JacekRuzyczka, 2024-12-22:
### Discussed in https://github.com/fdivitto/FabGL/discussions/317

<div type='discussions-op-text'>

<sup>Originally posted by **badre2911** January 18, 2023</sup>
Hi,
I need to print text file utf-8 on terminal.
Does the FABGL library support UTF-8 ?
Thanks.</div>

Background: For > 10 years, Linux systems have been communicating with text consoles in UTF-8 only, whereas the last "real" VTs appeared before the advent of UTF-8.

peter-kutak, 2025-05-01:
Hi, I made fork uFabGL with quick and very dirty modification for basic support UTF-8

JacekRuzyczka, 2025-05-08:
Good idea, @peter-kutak ! In fact, I'm now struggling with some flaws in the keyboard layout definitions.