fdivitto/FabGL #416: AnsiTerminal: Two vttest tests have failed.

JacekRuzyczka, 2025-06-25:
I've just run parts of the vttest suite against the AnsiTerminal program. Two of them failed:

The test for ANSI colors revealed that the terminal can't display the color blue, neither the bright, nor the "normal" flavor. Magenta is displayed as red.

[image]

The other test is called character position absolute (HPA), which should display a box of asterisks, but the left and right edge of the box are depicted the wrong way:

[image]

Terminal type is ANSI. I am using an Olimex board, but the guys from Olimex said they were not competent in this matter.

```
** FabGL — Serial Terminal                           **
** 2019–2022 by Fabrizio Di Vittorio — www.fabgl.com **

Version           : 1.6
Screen Size       : 640 × 480
Terminal Size     : 91 × 36
Terminal Type     : ANSI
Serial Port       : FabGL Terminal: TX=2 RX=34
Serial Parameters : 115200,8N1 flow=None
```

JacekRuzyczka, 2025-08-15:
@johngalt01  Which FabGL board are you using? THe original one from Fabrizio, or the Olimex one?

johngalt01, 2025-08-17:
its a LILYGO esp32 and i just flashed the eprom with the ASCI Terminal code

JacekRuzyczka, 2025-08-22:
I've also tried Space Invaders on the same board, and in this example, white graphics are displayed as such. Otherwise, they would have been yellow.

johngalt01, 2025-08-24:
you don't have to go that far, just connect to a host machine via a serial or usb connection then cat or type the .bmp file and it will output to the terminal and it will draw a picture for you.

its just like the .vt files here : http://artscene.textfiles.com/vt100/

the BMP files came from an early test i did a few years ago which was a proof on concept. i just needed to call them BMP files at the time but they are just txt files with escape codes in them.

if you have a CP/M or DOS machine and have the ansi terminal as your primary display then just have them type to the screen.

JacekRuzyczka, 2025-08-24:
OK, I've just tested the board with the `GraphicalUserInterface` example sketch, and believe me or not: It does display blue and cyan! 

[image]

Your example is to be used with the `AnsiTerminal` sketch on the board, right? Then I would have the same problems with blue color as previously, wouldn't I?

JacekRuzyczka, 2025-08-29:
Well, I've not at least got a hint on what might cause the problem with the almost missing blue color on Olimex boards: `AnsiTerminal` uses a [`BitmappedDisplayController`](https://github.com/OLIMEX/FabGL/blob/04f328a10573297dd554f13be7f369cdee0f7a2b/examples/VGA/AnsiTerminal/AnsiTerminal.ino#L37C8-L37C34) (which is possibly not working properly on Olimex boards), whereas `SpaceInvaders` uses a [`VGAController`](https://github.com/OLIMEX/FabGL/blob/04f328a10573297dd554f13be7f369cdee0f7a2b/examples/VGA/SpaceInvaders/SpaceInvaders.ino#L41C8-L41C21), and `GraphicalUserInterface` relies on a [`VGA16Controller`](https://github.com/OLIMEX/FabGL/blob/04f328a10573297dd554f13be7f369cdee0f7a2b/examples/VGA/GraphicalUserInterface/GraphicalUserInterface.ino#L34C8-L34C23). I can't find out more about this error, as the low-level code is written in Assembler. Maybe @peter-valkov could help us here.